orbingol/NURBS-Python #6: hohoho!!!

nortikin, 2017-05-10:
Great i found you.
Sverchok must have it onboard

orbingol, 2017-05-10:
@nortikin I wish I could understand what you mean, especially with that title "hohoho!!!". Are you trying to be funny or what?

nortikin, 2017-05-10:
i'm happy to find your repo.
Newermind. Thank you for your work

nortikin, 2017-05-10:
sorry, Sverchok is http://github.com/nortikin/sverchok

orbingol, 2017-05-10:
@nortikin You could have told these directly (or, maybe, emailed me) with a more relevant title and descriptive figures. I thought that somebody was spamming the issue tracker, which is always possible as this project is on a public domain and freely accessible. But, I got your point. Thank you for your messages and nice figures :)

I really appreciate and also respect other people's work on computer graphics, as I believe that all of these works are results of incredible amount of time and effort. I take every issue and every comment on Github very seriously and I try to make everything right as much as possible. I also receive emails on some issues, and subsequently fix them (or rewrite the code causing the issue). I really appreciate people testing and using my code for their own research or other purposes.

Thanks again for your time.

closed 2017-05-10 by orbingol

nortikin, 2017-06-19:
can this tool be in numpy? or don't you know who maybe made numpy version?

orbingol, 2017-06-19:
@nortikin I would answer your first question as yes, but no :)

Well, actually it is impossible to underestimate the advantages of numpy package in programming mathematical tools, but I believe that people use numpy a lot even it is unnecessary to use in some development environments. It would be very advantageous to use numpy in this NURBS package, but I believe that if I could do it with minimum dependencies, I should do it like that.

However, I think you might want to ask for a feature which returns a numpy array instead of a 2D native list structure. That is something doable but it should be optional, so that people who are not using numpy package can use this NURBS package too. Not everyone is using the well-known Python distributions (including me) and in some development environments, it might be impossible to install/compile some of the dependencies (I have seen some).

I am very open to developing new features that people can advantage of. If you could open **a new feature request** with all possible details (and with some reasons on why we should add it to this package), I would try to implement as soon as possible.