orbingol/NURBS-Python #7: Implement Knot Insertion

orbingol, 2017-05-11:
Knot insertion is one of the fundamental geometric algorithms stated in _The NURBS Book_.  The major advantages of the knot insertion operation are;
* Splitting (subdividing) curves and surfaces
* Interactive design (adding control points to make shape control more flexible)

Let's start implementing the knot Insertion algorithms (A5.1 for curves, A5.3 for surfaces).

rainman110, 2017-07-03:
If you still want to implement knot insertion, here is the code from our python based nurbs library:

```python
    def insert_knot_single(self, x):
        """
        Inserts a knot x into the b-spline. x must be inside range (knots[0], knots[end])

        Args:
            x (double): knot value to insert into b-spline

        Returns:
            Index of inserted knot in knot vector
        """

        if self.is_rational:
            return self.wrap_nurbs__('insert_knot_single', x)

        if x <= self.knots[0] or x >= self.knots[-1]:
            raise Exception("Knot value is outside valid knot range")

        # count multiplicity of knot
        multiplicity = np.count_nonzero(np.abs(self.knots - x) < 1e-14)

        pos = find_pos(self.degree, self.knots, x)
        if multiplicity == self.degree:
            # knot has full multiplicity already, stop
            return pos

        self.cp = np.hstack((self.cp[:, 0:pos + 1], self.cp[:, pos:]))

        k = np.arange(pos - self.degree + 1, pos + 1)
        alpha = (x - self.knots[k]) / (self.knots[k + self.degree] - self.knots[k])
        self.cp[:, k] = alpha * self.cp[:, k] + (1. - alpha) * self.cp[:, k - 1]

        self.knots = np.hstack((self.knots[0:pos + 1], x, self.knots[pos + 1:]))
        return pos + 1
```

Feel free to use and adapt it for your needs.

orbingol, 2017-10-14:
This feature is implemented but I ran into some other compatibility/integration issues with the other libraries I have been using/developing, so all fixes/improvements + knot insertion (and more if I could find time to test them) will be released in the next major release, v3.0

closed 2017-10-14 by orbingol