orbingol/NURBS-Python #19: Wrong exception wording for Curve2D

goatchurchprime, 2018-01-23:
```python
import geomdl.BSpline
curve = geomdl.BSpline.Curve2D()
curve.ctrlpts = [(1,2,3)]

```
----> ValueError: ERROR: Please input 3D coordinates

Don't you mean "Please input 2D coordinates?"

orbingol, 2018-01-23:
Thanks for the heads up @goatchurchprime 

I just pushed the fix to `devmaster` branch. It will be fixed on the next release.

goatchurchprime, 2018-01-23:
Just a note: I had a lot of problem with this library because somehow I got v2 and v3 installed on my computer, while a colleague I was developing the software for only had v3 on his.  

So "import nurbs" worked on my computer, but did not on his.  
https://github.com/goatchurchprime/dxf2svg/blob/master/dxf2svg/dxf2svg.py#L15

A real bugger to discover this problem because I didn't know anything of the invisible change to "import geomdl", and my v2 library was not responding to pip unistall, until I deleted it as a directory.  

I wanted to upgrade to v3, but it appears that you don't have a setter for geomdl.NURBS.Curve2D.ctrlpts.  Is this correct?

orbingol, 2018-01-23:
From what you wrote, I feel that I have to explain some things.

I would like to answer your complaints on the name change. I thought that I have clearly indicated the name change on the documentation (http://nurbs-python.readthedocs.io/en/latest/modules.html). I am truly sorry for the inconvenience that the name change caused.

Since, I have received some other complaints, such as "NURBS-Python is overwriting somebody else's existing work on NURBS because that somebody else named his module as `nurbs`", which I believe I am not responsible but others think I am the guilty one, I had to take the liberty to change the module name to `geomdl`. I know this is a big mistake but I have been left with no chances. I even googled `geomdl` first before I changed it last year, which shouldn't even be necessary.

Just to make it clear for the future releases, there will be no name changes and I will stick to 'geomdl' no matter what it causes to others.

In any case, you could have asked here about the problem (or emailed me). I am trying my best to reply to emails regarding to NURBS-Python as soon as possible, and I even reply to the funny and unreasonable ones with all possible respect. Just wanted to keep this as a note here.

I believe package not responding to pip uninstall is an python issue. First of all, installing with pip and setup.py works in a different way. So, if you install a module with setup.py, you cannot uninstall it with pip (this is how I observed after installing and uninstalling hundreds of times, but I still could be wrong). Secondly, when you install the package using `setup.py install`, there seems to be no way to uninstall, or let's say I couldn't find a way to make `setup.py uninstall` works in the most portable way. As long as I remember, I have to store the list of files somewhere, maybe as a variable, and then create an uninstall function, then try to find files and delete and so. I don't feel this is a correct way to uninstall a module and I believe the things should be changed on the setuptools side. I am still sorry for the issues that you faced, but this is all I could do.

Finally, I would like to answer your last question about the setter. There is a setter but it seems it is not working as intended. I will take a look at it.

orbingol, 2018-01-23:
I opened #20 for the setter issues.

closed 2018-01-23 by orbingol

goatchurchprime, 2018-01-24:
Thanks.  I hadn't expected quite such a long reply, but you hit on all the unfortunate events I encountered.  The other critique of the module system is there's not a lot of connection between the X-values of "pip install X" and "import X", so I didn't even get as far as tracking down the docs for this nurbs library and was being confused with something entirely else.

orbingol, 2018-01-24:
You are absolutely right about the confusion between "NURBS-Python" and "geomdl" if you consider the pip side. Actually, I have another long story behind it, but, yeah, it is long :) After all, it is already too late to correct it I guess, especially after publicly announcing that `geomdl` is final :)

Just to eliminate confusion, I created a repository with some examples (here: https://github.com/orbingol/NURBS-Python_Examples) and put the results in the documentation (here; http://nurbs-python.readthedocs.io/en/latest/visualization.html). I always advertise the source code, examples and the documentation together.

commit 1164236b (2018-01-23): Fixes #19