orbingol/NURBS-Python #21: Linearize curves and surfaces to a tolerance

goatchurchprime, 2018-01-25:
It looks like curvepts is made by proceeding at a sample rate of _delta from 0 to 1 proportionally along the whole length of the spline.  

For use in CAM we want the geometry subdivided only as far as necessary that the the approximation fits the the curve within the margin of a tolerance value.  Often we will set this to the tolerance of the CNC machine, but we don't want to overdo it, because the controllers tend to choke on too many close together points.

orbingol, 2018-01-31:
Would using a small delta work for you? You can set is using the `delta` property.

The figure below generated by using a delta value of 0.1

[image]

goatchurchprime, 2018-01-31:
No.  Just setting a small sample rate generates too many unnecessary segments.  This is a big problem where the nurbs are very flat and straight and your CNC machine chokes on these points when it is built to run smoothly along what is a straight line. 

As I said, the optimum for CAM (be it laser cutting or machining) is as few straight segments as possible, but within the deviation tolerance of the desired curve.  

A cheap way to implement this is to oversample the curve at a high sample rate, and then delete points that are almost colinear with their neighbours until the curve moves too far away from the original.  

However, if you have the maths right in terms of understanding the curvature of the nurbs, you can directly calculate the deflection of the curve from the straight line segment between two sample points, and therefore produce exactly the right sample rate along any nurbs curve to maintain tolerance for the resulting piecewise linear approximation.

orbingol, 2018-01-31:
It seems super easy to implement and definitely there is no need to change any single line of NURBS-Python. The implementation could be even done in a separate package which extends NURBS-Python, and I will definitely advertise it on the README file.

Because I am not currently working on CAM topics and I have extremely limited time due to my PhD research,  I can only start working these next year (yes, 2019 and most probably some time close to the end of 2019) and I cannot promise for anything.

Regarding to that, any help would be very much appreciated. As I said before, you don't even have to change any single line of the existing NURBS-Python package to implement changes.

closed 2018-05-05 by orbingol