orbingol/NURBS-Python #26: Utilities - Wrong number of knots

wolfysong, 2018-04-19:
function generate_knot_vector may generate wrong number of knots

Original code:
    # Middle knots
    knot_vector += [mid_knot for mid_knot in frange(0, 1, spacing)]

number of Middle knots can be one more than expected. A quick fix:
    # Middle knots
    knot_vector += [mid_knot for mid_knot in frange(0, 0.99999999999, spacing)]

orbingol, 2018-04-19:
@wolfysong thanks.

It is because of the floating point error caused by division inside the `frange` function. It seems that  python rounds the floating points to the nearest integer value while using `<` and `>` operators, even the floating point is in python's precision range.

What I observed is that it sometimes incorrectly determines the result for less than and bigger than operators when the floating point number has 5 fractional digits on a 64bits python distribution regardless of the library or the wrapper used.

I also observed the same problem in determining `delta` from `sample_size` property. If the sample_size = 50, then `frange` fails to generate the last knot (or the set of last rows - columns) for the evaluation. If sample_size = 51, it just works fine.

I am completely open to the new `frange` implementations (without using Numpy) or some ideas on fixing the floating point error.

orbingol, 2018-04-21:
@wolfysong could you check this function with your data? This is from v4.x of NURBS-Python (still work in progress), I haven't had enough time to test thoroughly but it should work, at least till I figure out some solution for `frange` function.

https://gist.github.com/orbingol/76d9f52f1c90afb5d2d201eb4ad14126

houtao-tud, 2018-04-25:
Hi, orbingol. I'm wolfysong's phd student, we have update the NURBS-Python package to 4.x-dev.
1: we try to read a txt document with 3000 control points, it shows: ValueError: Input is not a valid knot vector;
2: we decrease the control points to 2500, 2000, 1500, 1000 ,500, it displays the same error, but it successed when 250 control points;(we attached the txt document with 3000 control points
[control points.txt](https://github.com/orbingol/NURBS-Python/files/1946297/control.points.txt)
)
3: we delete the line 122 and 123 in BSpline.py, the program run success with 3000 control points.

houtao-tud, 2018-04-25:
@orbingol I have checked the code on the master branch, but met the same problem.

orbingol, 2018-04-25:
Interestingly, it is working for me. Could you please double check that you have all the updates on the master branch?

[image]

houtao-tud, 2018-04-25:
@orbingol Thanks for your kind help, at first i haven't updates to the master branch, i only try the example code of master version. i have update it again and it is OK now. it is so nice that i can go on my research.

orbingol, 2018-04-25:
@houtao-tud thanks for using NURBS-Python and reporting this bug. I have also released a new version on PyPI which contains the updated files.

commit 1e1f75ba (2018-04-25): Update knot vector generator function - fixes #26
closed 2018-04-25 by orbingol