orbingol/NURBS-Python #28: Add support for trimmed surfaces

orbingol, 2018-05-05:
The title explains it all.

orbingol, 2018-10-01:
v4.3.0 will be released with basic support for trimmed surfaces. Please see #29 for tessellation algorithms.

BinWang0213, 2018-10-10:
Hi,

Is there any plan or idea to support surface-surface intersection? I think this Boolean operation somehow related to trimmed surfaces.

Thanks for your nice code,
Bin

orbingol, 2018-10-10:
@BinWang0213 thank you very much for your question.

Let me explain what I understand from surface trimming. To trim a surface, we introduce closed curves on the parametric space of the surface. These curves can be NURBS - B-Spline curves or analytic curves (circle, square, rectangle, etc.) and they are called trimming or trim curves. Since the trimming curves are 2-dimensional and trim curve points correspond to (u,v) parametric values of the surface, we can easily evaluate the trim curves on any surface with any shape. The reference (or one of the references) for this method is the paper authored by Piegl and Richard (1995). There could be some other methods but my plan is implementing this method.

To my understanding, surface-surface intersection requires computation of the intersection curves on the 3-dimensional space. Intersection curves can trim the surfaces if they have the required conditions. There will be support for computation of intersection curves in NURBS-Python but it is a long term plan.

BinWang0213, 2018-10-11:
Thanks for you rapid reply. Looking forward to see the updates in the future.

closed 2019-02-16 by orbingol

yijiangh, 2020-10-06:
> To my understanding, surface-surface intersection requires computation of the intersection curves on the 3-dimensional space.

Out of curiosity, what is the standard method for computing this intersecting curve in 3D? I suspect we need to solve a set of algebraic equations to do this since this curve is a NURBS - B-Spline curves on both given surfaces?

gilesgs, 2024-02-22:
First comes the tolerance since NURBS are usually not exact then like  all good things in life a ,matrix must be inverted

commit 66a57424 (2019-02-06): Add trimming algorithm - closes #28