orbingol/NURBS-Python #34: Problems in evaluation of derivative control points

orbingol, 2018-07-29:
From #33 @Nodli says

> I think I found a bug in the computation of weights in derivative_ctrlpts for NURBS Curves and Surfaces.
> I am currently trying to solve this.

orbingol, 2018-07-29:
@Nodli would you elaborate a little on the bug you have found?

orbingol, 2018-07-29:
Issue #35 could also be related to this one.

Additionally, `derivatives_ctrlpts` fails with the error "division by zero" when there are repeated knots in the knot vector. Trying to fix division by zero error caused derivative function to fail. I started to think that The NURBS Book algorithm A3.3  designed to support algorithm A3.4 (derivative algorithm) but not to be used as stand alone functionality.

Nodli, 2018-07-29:
This is the issue I was talking about.
I don't think this is a bug after looking into it but it's related to the method.
The last two equations of p.125 are exactly what we are doing and what is expected: when two subsequent weights are equal, the weight of the derivative control point is zero which raises the "division by zero" error when trying to evaluate it to non-rational coordinates.
I did not find a solution but there must be one as at the begining of section 4.3 p.125 the book says "In Section 3.3 we developed formulas and algorithms to compute the derivatives of nonrational B-Splinecurves. Those formulas and algorithms apply, of course, to Cw(u) [...]"

orbingol, 2018-07-30:
From what I understand from chapter 4, I can say that they are computing C(u) and S(u,v) from Cw(u) and Sw(u,v). Well, simplifying everything to this only sentence may not be that correct but the authors (and the majority of the other researchers) use weights as shape modifications tools, like a supplement to what we define with (x,y,z). This also correlates to the main purpose of the "rational" curves and surfaces.

Regarding to what you say about the beginning of Section 4.3 on p.125, I feel a small trick. Yes, it is definitely possible to differentiate Cw and Sw with the methods developed in Section 3.3 (and I depend on this sentence on implementation of Algorithm A4.2 RatCurveDerivs and it just works) but not with all of them, for instance A3.4. Sometimes it is just the style of writing.

As a result, this is not a bug at all but the method itself. Maybe we could elaborate more on the Issue #35 and try to find an algorithm for that. What do you think about it?

Nodli, 2018-07-30:
I did some research on this today and it seems you were right.

Details about what i am saying are in this paper "Evaluation and properties of the derivative of a NURBS Curve", S. Floater (https://pdfs.semanticscholar.org/23cf/5b46ea38dd321c525bf952bde459dbf1be33.pdf).

The equation from the book comes from the "hodograph formula" for non-rational Bezier and BSplines (Eq (5) p.6 in the paper) by identifying the evaluation formula in this one.
However when adapting the hodograph formula for rational curves(Eq (14) p.8), either Bezier or BSplines, more terms are related to the basis functions and their derivatives and I don't think it is possible to identify the evaluation formula.

Unfortunately I could not find any other method for rational curves...

orbingol, 2018-07-30:
Thanks @Nodli for the research.

So it turned out that there are no bugs and I guess we can close this issue.

closed 2018-07-31 by orbingol