orbingol/NURBS-Python #37: Cannot export evalpts

nikorose87, 2018-12-17:
Hey buddy,

I assume you have a bug trying to export the `evalpts`, in the function `_export_txt` you have the following:

`
    else:
        # B-spline or NURBS?
        try:
            ctrlpts = obj.ctrlptsw
        except AttributeError:
            evalpts = obj.evalpts 
            ctrlpts = obj.ctrlpts
        # Loop through points
        for pt in evalpts: # instead ctrlpts
            line = sep.join(str(c) for c in pt) + "\n"
            fp.write(line)
`
I modified the lines so that the function reads the evalpts to write instead the control points. Otherwise you will write the ctrlpts always.

orbingol, 2018-12-17:
Well, `exchange.export_txt` was never intended to export evaluated points but only control points. We needed it in that way for some compatibility reasons with the existing libraries we use.

On the other hand, you can use `export_csv` to export evaluated points and the control points. There is a minor difference between `export_csv` and `export_txt` and therefore, these functions were refactored heavily on the `devmaster` branch.

orbingol, 2018-12-18:
I now understand what you mean. It seems that the following API call on v4.4.1 returns the control points instead of evaluated points.

```
exchange.export_csv(ns, "exported_evpts.csv", point_type="evalpts")
```

I guess for now, you can safely downgrade to version v4.3.8 temporarily. I have started updating exchange module on v4.4.x series and that should be the problem.

Thanks for reporting the bug :) I will release an updated version (with the proper unit tests) as soon as possible.

orbingol, 2018-12-18:
Would you test it again with the version 4.4.2 @nikorose87?

nikorose87, 2018-12-18:
The issue was solved. Thanks

orbingol, 2018-12-18:
Great :) Thanks again for reporting the bug.

closed 2018-12-18 by orbingol