orbingol/NURBS-Python #38: Any Pure-Python CSG/CAD lib good interoperability ?

nestarz, 2018-12-17:
Hello @orbingol and others !

Thanks very much for your work, hope to contribute !

I'm building a pure-python serverless (AWS Lambda) STL object generation tool for creative coding, and I was wondering if you knew if NURBS is intended to do CSG operations or if there is a good library that is also pure-python and works well with NURBS-Python.

Thanks again,
Elias

orbingol, 2018-12-18:
Hi Elias,

Your project looks really interesting. I think it is possible to use NURBS in creative coding but depending on what you plan to do, the amount of work might change. What I understand from CSG operations is "Boolean Operations", such as union and intersection. Here, it might be good to say that there are some differences between parametric modeling (NURBS) and solid modeling (BRep, CSG). A good and complete CAD software should utilize all these techniques together to provide efficient modeling. For instance, it is definitely easier to apply some geometrical operations in CSG but most of the time it is more convenient to use BRep in solid representations.

Some of the boolean or boolean-like operations can be easily implemented in the NURBS way; for instance knot insertion for splitting and decomposition, knot removal to join adjacent NURBS shapes (curves, surfaces and volumes) and similar. On the other hand, some of the boolean operations could be a little hard to implement, such as computing surface-surface intersections or more.

The current development version of NURBS-Python has some of the fundamental algorithms, such as knot insertion and fitting, and I will add knot removal, degree reduction and degree elevation for curves and surfaces in the upcoming months. However, currently I have no plans to work on algorithms like surface-surface or surface-volume intersections.

If we are considering CSG directly, I remember playing with [CSG.js](https://evanw.github.io/csg.js/) and I remember finding a Python version of it on PyPI but it looked abandoned to me at that time. These are not guaranteed to work with NURBS-Python, though.

Thanks,
Onur

orbingol, 2019-01-16:
I am closing this issue now. If you need to discuss anything, feel free to reopen it @nestarz

closed 2019-01-16 by orbingol