orbingol/NURBS-Python #39: CAD File import for Blender

bunterWolf, 2018-12-21:
Hi orbingol,

Commercial 3D programs like Cinema 4D and Maya can import, tessellate and render CAD data. Unfortunately [Blender](https://www.blender.org/) does not have this possibility. But I believe that this will be of great importance for this software. That's why I'm trying to figure out how something could be made possible in this direction. Thereby I came across Nurbs Python.

Do you believe your Nurbs Python library could make it possible for Blender to read and tessellate CAD files in order to create high-quality renderings?

Thank you very much for your time
Sebastian

bunterWolf, 2018-12-23:
Thank you very much for your detailed answer. :)
I expected something in this direction and I will probably accept that a mesh from a iges or step is not to get. I have to limit myself to obj, fbx and 3ds for further steps.

______
My thoughts only for you, if you find it interesting in general ;)

I have seen that you are working on an IGES import and there is already an OBJ export. My thought was that you could use Python Nurbs to create a mesh to render from an IGES file. 

But the way you describe it... I guess I imagined it to be too easy. :)

I named it openly as a CAD file, because all the file types have advantages and disadvantages. There simply isn't the right file format.

Can Blender already:
FBX: materials, Rhino and Fusion 360 but no Solidworks
OBJ: materials, only Rhino
STL: [No good rendering quality](https://blenderartists.org/t/3d-model-issues-after-stl-import-with-smooth-shading/702684), no materials

CAD standards:
IGES: Super compatibility, old format with problem with complicated bodies. I don't believe in materials.
STEP: Best compatibility, but probably the most complicated. I don't believe in materials.
(If this would work, you could adjust the mesh details in Blender how you need them.)

Rhino 3ds: [Already contains the mesh, Blender importer is in development](https://github.com/jesterKing/import_3dm)
Solidworks format: no idea, probably not even open
Fusion 360: No idea, probably not even open

STEP or Iges would be the most flexible, but well tessellation... complicated :X

Comprehensive CAD import concept: [Unreal Datasmith.](https://docs.unrealengine.com/en-us/Studio/Datasmith/Overview) But here performance and polycount are of course super important and it is very effortful implemented here. A bit over to top for the beginning :)

orbingol, 2018-12-24:
@bunterWolf thank you so much for the links :) They will be good reference for me and the others.

Unfortunately, I don't think I will be able to find time for developing IGES (#31) or X3D (#24) importer/exporter for the next release of NURBS-Python. It is still possible to use external libraries for importing and exporting these formats (like I do for .3dm format); however, there are no shortcut API calls added to the code at the moment. Although importer and exporters are very important, I would like to focus on the algorithms and the visualization part for the next release.

I would be really happy if somebody else could pick these tasks and implement these importers and exporters for NURBS-Python. I don't expect a PR, that might be a little bit too complicated but an add-on module will also be good and I can also advertise such modules on the README file (just let me know about the modules).

bunterWolf, 2018-12-25:
I'd like to go deeper, but I'm a designer with some programming skills. The whole mathematics is unfortunately too advanced for me...
I expect that the Step topic in Blender will still exist for another 2 years, maybe something has come up by then. 

Thank you very much for your time and your tips! :)

closed 2018-12-25 by bunterWolf