orbingol/NURBS-Python #41: VTK renders are very noisy compared to MPL renders

abhishekpandey07, 2019-01-25:
**Describe the bug**
The VisVTK Visualisation module renders partly deformed surfaces. VisMPL module works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. fit a point cloud.
2. set delta to 0.02
3. render using VTK
4. See error

**Expected Behavior**
The VisVTK module should render the surfaces as they are.

**Configuration:**
 - OS: Ubuntu 18.10
 - Python distribution: Anaconda
 - Python version: 3.6.8
 - geomdl install source: Anaconda
 - geomdl version/branch: 5.0b4

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]
[image]

orbingol, 2019-01-25:
The reason for this problem is that geomdl MPL renderer uses a custom tessellation algorithm instead of Matplotlib's integrated QHull library but geomdl  VTK renderer uses the default Delaunay triangulation filters which come with the VTK library itself. It is also possible to observe a similar error when a cylindrical or a conical surface is rendered.

Actually, I wouldn't consider this issue as a bug of geomdl but maybe an integration problem related to how different renderers work. I was able to solve it by using a custom tessellator for Matplotlib. If there is a way to feed pre-generated triangles to VTK system (definitely without using their Delaunay triangulation filter), then I guess, that would be the solution.

abhishekpandey07, 2019-01-25:
Alright, Thanks for making that clear.

orbingol, 2019-03-11:
I've pushed a fix on `devmaster` branch and tested with a couple of multi-surface objects. It seems to work fine.

orbingol, 2019-03-22:
This issue is fixed with the release of geomdl v5.2.0

closed 2019-03-22 by orbingol