orbingol/NURBS-Python #42: Can't Import Fitting Module

Mitch981521, 2019-02-21:
**Describe the bug**
A clear and concise description of what the bug is.

After installing NURBS-Python using pip, when I try to import the "fitting" module, I get an error saying "cannot import name fitting"

**To Reproduce**
Steps to reproduce the behavior:
1. Execute "pip install geomdl" install command
2. Copy or download the following example from NURBS-Python_Examples:   NURBS-Python_Examples/fitting/interpolation/global_surface.py
3. Run the File

**Expected Behavior**
I expected the file to run with a visualization of the approximated surface

**Configuration:**
 - OS: Ubuntu 16.04
 - Python distribution: OS default
 - Python version: 2.7.12
 - geomdl install source: "pip install geomdl" command
 - geomdl version/branch: 4.4.4

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
StackOverFlow says that it may be due to circular dependencies?

orbingol, 2019-02-21:
Fitting module is not available in v4.x. Try installing the latest beta version of v5.x, e.g. `pip install geomdl==5.0b7`

You can find the list of releases here: https://pypi.org/project/geomdl/#history

orbingol, 2019-02-22:
I've released geomdl v5.0.0 today. `pip install geomdl` should install the latest version correctly.

closed 2019-02-22 by orbingol