orbingol/NURBS-Python #43: Typing Module Not Found

Mitch981521, 2019-02-22:
**Describe the bug**
After installing NURBS-Python using pip, when I try to import the "fitting" module, I get an error saying

" from geomdl import fitting
    from typing import Any, Sequence, List, Tuple
ImportError: No module named typing"

**To Reproduce**
Steps to reproduce the behavior:
1. Execute "pip install geomdl" install command

2. Copy or download the following example from NURBS-Python_Examples:   NURBS-Python_Examples/fitting/interpolation/global_surface.py

3. Run the File

**Expected Behavior**
I expected the file to run with a visualization of the approximated surface.  

**Configuration:**

* OS: Ubuntu 16.04
* Python distribution: OS default
* Python version: 2.7.12
* geomdl install source: "pip install geomdl" command
* geomdl version/branch: 5.0.0

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
Website says that NURBS is compatible with python 2.7.x and 3.4.x.  I'm not able to find "Typing" module under python 2 documentation, so will I need to upgrade python?

orbingol, 2019-02-22:
@Mitch981521 you don't need to update Python because Ubuntu 16.04 comes with Python 3 too (via `python3` and `pip3` commands). Since Python 2 will be deprecated in January 1st, 2020 ([ref1](https://pythonclock.org/), [ref2](https://www.python.org/dev/peps/pep-0373/#id4), [ref3](https://python3statement.org/)), I would suggest using Python 3 if you have no specific Python 2 dependent modules.

geomdl will support Python 2.7 for a little bit longer than January 1st, 2020, let's say till `pip` stops working for Python 2.7.

If we return back to the problem, it looks like there is an issue with the packaging code that I have been using for a while. It should be fixed with geomdl v5.0.1. Please update to the latest version.

orbingol, 2019-02-28:
I think this issue has been resolved. Let me know if experiencing any problems.

danielj195, 2019-03-01:
I've been using it on a virtual environment because I was unaware that I had python3.  No issues so far!

Mitch981521, 2019-03-01:
@orbingol If there is a way to make it work with Python 2, that would be greatly appreciated.  We are wanting to use this package as part of a program that will be run on ROS (robot operating system), which  unfortunately is not yet implemented for Python3.

orbingol, 2019-03-02:
@Mitch981521, I couldn't understand the problem. Are you still having the same issue? @danielj195  says it is working fine? Because, I was able to run it on Ubuntu 16.04 with Python 2.7.12.

If we go into the details, you were having a very simple issue which could be solved by running `pip install typing` but it should have been installed with the geomdl package. I fixed the packaging issue and it now installs the modules correctly.

Would you please answer the following questions? I'd like to track down the exact error, if possible:

1. Have you tried running `pip install geomdl --upgrade` as instructed [here](https://nurbs-python.readthedocs.io/en/latest/install.html#install-via-pip)?
1. Just in case, have you tried running `pip install typing`?
3. After trying 1 and 2, are you still getting errors?
4. If you are getting errors, could you please write them down?

Mitch981521, 2019-03-11:
Hi @orbingol , sorry for the late reply.  I used pip to install typing and got it to work.  Thanks!

closed 2019-03-11 by orbingol

markrazz, 2019-12-08:
I get this same issue
I have tried everyhting

!pip install --user C:\3DCode\NURBS-Python-master
Processing c:\3dcode\nurbs-python-master
Building wheels for collected packages: geomdl
  Running setup.py bdist_wheel for geomdl: started
  Running setup.py bdist_wheel for geomdl: finished with status 'done'
  Stored in directory: C:\Users\aura\AppData\Local\pip\Cache\wheels\89\dd\90\740280006b3d633e4657c325d0b80b623de3b56f202957c007
Successfully built geomdl
Installing collected packages: geomdl
Successfully installed geomdl-5.2.9
twisted 18.7.0 requires PyHamcrest>=1.9.0, which is not installed.
grin 1.2.1 requires argparse>=1.1, which is not installed.
You are using pip version 10.0.1, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
  File "C:/!python/NurbsPy.py", line 17
    from geomdl importBSpline
                            ^
SyntaxError: invalid syntax

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
Traceback (most recent call last):

  File "<ipython-input-16-3ec3d3973a88>", line 1, in <module>
    runfile('C:/!python/NurbsPy.py', wdir='C:/!python')

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 668, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 93, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/!python/NurbsPy.py", line 17, in <module>
    from geomdl import BSpline

ImportError: No module named geomdl

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
  File "C:/!python/NurbsPy.py", line 18
    from C:\3DCode\NURBS-Python-master\geomdl import BSpline
          ^
SyntaxError: invalid syntax

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
  File "C:/!python/NurbsPy.py", line 18
    from C:\3DCode\NURBS-Python-master\geomdl import BSpline
          ^
SyntaxError: invalid syntax

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
  File "C:/!python/NurbsPy.py", line 18
    from C:\3DCode\NURBS-Python-master\geomdl import BSpline
          ^
SyntaxError: invalid syntax

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
Traceback (most recent call last):

  File "<ipython-input-20-3ec3d3973a88>", line 1, in <module>
    runfile('C:/!python/NurbsPy.py', wdir='C:/!python')

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 668, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 93, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/!python/NurbsPy.py", line 18, in <module>
    from geomdl import BSpline

ImportError: No module named geomdl

pip install geomdl --upgrade
  File "<ipython-input-21-0219797c7932>", line 1
    pip install geomdl --upgrade
              ^
SyntaxError: invalid syntax

!pip install geomdl --upgrade
Requirement already up-to-date: geomdl in c:\users\aura\appdata\roaming\python\python27\site-packages (5.2.9)
twisted 18.7.0 requires PyHamcrest>=1.9.0, which is not installed.
grin 1.2.1 requires argparse>=1.1, which is not installed.
You are using pip version 10.0.1, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

runfile('C:/!python/NurbsPy.py', wdir='C:/!python')
Traceback (most recent call last):

  File "<ipython-input-23-3ec3d3973a88>", line 1, in <module>
    runfile('C:/!python/NurbsPy.py', wdir='C:/!python')

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 668, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda2\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 93, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/!python/NurbsPy.py", line 18, in <module>
    from geomdl import BSpline

ImportError: No module named geomdl

orbingol, 2019-12-09:
It looks like a completely different issue and I think it is not related to `geomdl` at all. It could be a `PYTHON_PATH` problem or something else. Please refer to the following doc page for help:

https://nurbs-python.readthedocs.io/en/latest/q_a.html#help-and-support