orbingol/NURBS-Python #45: Vertex data faulty conditional?

Mitch981521, 2019-03-11:
**Describe the bug**
When I am importing my data using the NURBS module, the data setter in elements.py appears to unnecessarily give the following error:

raise ValueError("Vertex can only store 3 components")

**To Reproduce**
Steps to reproduce the behavior:
1.  Put the "nurb_vertex_bug" and data file into the same directory.
2.  Modify the path to the data file in the import function in the "nurb_vertex_bug" file
3.  Execute "nurb_vertex_bug"

**Expected Behavior**
I expected a successful rendering of a NURBS surface corresponding to this data.  Here are the steps I've already taken to try to resolve this issue:

-Changed type of input data (tried both 2D list and list of tuples)
-Verified using a for loop that each element of this list does in fact have length of 3
-Looked at the source code for possible clues as to what is causing the ValueError

**Configuration:**
 - OS: Ubuntu 16.04
 - Python distribution: Default
 - Python version: 2.7.12
 - geomdl install source: Pip install 
 - geomdl version/branch:  5.1.2

[nurb_bug.zip](https://github.com/orbingol/NURBS-Python/files/2950661/nurb_bug.zip)

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
I am using a lot of data, so maybe it has to do with the way my data is formatted?  It just seems like the error that is being reported does not match what is actually causing the error to be thrown.

orbingol, 2019-03-11:
The library follows the notation described in The NURBS Book by Piegl and Tiller. A NURBS surface accepts weighted control points by default as the control points input.  The NURBS Book calls it `Pw` which simply corresponds to a list of `(x * w, y *w, z * w, w)`. On the other hand, A B-Spline surface accepts control points in the format `(x, y, z)` and The NURBS Book calls it `P`.

Since you are inputting 3-dimensional set of points to a NURBS surface object, the last element is considered as the weights. It has to be 4-dimensional or simply use a BSpline surface object.

Mitch981521, 2019-03-11:
thank you for the prompt reply!  that makes sense, thank you

closed 2019-03-11 by orbingol