orbingol/NURBS-Python #47: Questions about Derivative Evaluation

Mitch981521, 2019-03-15:
**Describe the bug**
Similar as before, this may be more a question.  I'm needing to evaluate the partial derivative of a surface at a given point.  When I use the "derivatives" evaluator for surface, the output I get doesn't appear to match the form showed in The NURBS Book and the values do not match what I'm expecting.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a test surface (mine is too large to share) and call the derivative evaluator.  For the "order" argument, I used 2.
2.  The output produces a 3-dimensional list of numbers that don't make sense instead of the expected 2-dimensional list.

**Expected Behavior**
A two dimensional list of partial derivatives.  I'm confused as to why I'm getting a 3-dimensional list.  The NURBS book shows that the list should be 2-dimensional.

**Configuration:**
 - OS: Ubuntu 16.04
 - Python distribution: OS default
 - Python version: 2.7.12
 - geomdl install source: 
 - geomdl version/branch: 5.1.2
[image]
[image]

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
Add any other context about the problem here.

orbingol, 2019-03-15:
I am pretty much sure that the derivatives are working as accurate as it could possible and I don't think you mean to say that derivatives are not working. If so, please provide a proof showing the exact error using a comparable floating point precision.

Regarding to your question, I was wondering if you have checked the documentation. I have a strong feeling that you will find the answer there: https://nurbs-python.readthedocs.io/en/latest/module_bspline.html#geomdl.BSpline.Surface.derivatives

Moreover, you can use **nurbs-python@googlegroups.com** for general support and help. If you would like to use the issue tracker, that would be okay too but I would suggest using "Open a regular issue" link right below the Bug Report button. Here is the direct link: https://github.com/orbingol/NURBS-Python/issues/new

Mitch981521, 2019-03-15:
I did check the documentation (even the pdf online) and it says that SKL[0][1] corresponds to the first derivative with respect to v, SKL[0][2] second derivative wrt v, etc.  I'm more confused why each of the elements in SKL is a list....shouldn't the partial derivatives just be floats?  The page I showed above shows an element of SKL being assigned a float, so that's also why I'm confused.  Do you know of a place where I can find some examples?

orbingol, 2019-03-16:
I understand why you are getting confused. Let me answer your questions separately.

> A two dimensional list of partial derivatives. I'm confused as to why I'm getting a 3-dimensional list. The NURBS book shows that the list should be 2-dimensional.

`SKL` is actually a 2-dimensional list of `h` values where `h` is a 3-dimensional list, corresponding to the results of the derivative operation.

>  I checked the documentation (both the website and pdf) and it says that SKL[0][1] corresponds to the first derivative with respect to v, SKL[0][2] second derivative wrt v, etc. What I'm confused about is why each of the elements of SKL is a list....shouldn't the partial derivatives just be floats? The page of the NURBS book I showed above shows an element of SKL being assigned a float.

I think you are completely forgetting that you are trying to compute a derivative of a 3-dimensional point and you also have a bug in your statement. I understand that you assume the derivative is a float because The NURBS Book says something like `SKL[k][l] = 0.0`, right? I think you have the book, so could you please take a look at the the Algorithm A3.5 on page 103? On line 10 of A3.5, the authors are setting `S = 0.0` where `S` corresponds to the 3-dimensional Cartesian coordinates of a surface point at evaluated at the parameter (u,v). From what I understand from your question, you are okay with the computed surface points. From your point of view, a surface point should be a scalar a value too, but it is not. Likewise, `temp` variable of the same algorithm is set to a float value but then it is multiplied by `P` and `P` is the 3-dimensional (or 4-dimensional, if rational) control point. Please note that these are the algorithms, not the actual implementations and they can be quite different, depending on your choice of programming language.

> I'm not sure what the third dimension of SKL corresponds to.

By saying this, I think you were expecting a derivative on the parametric space of the surface but not on the Euclidean space. However, the algorithm you are referring to computes the derivative of `S(u,v)` which corresponds to a 3-dimensional point on the Euclidean space (page 111, right above Algorithm A3.6).

A surface is a mapping between a 2-dimensional parametric space to 3-dimensional Euclidean space. A surface point is a 3-dimensional point evaluated at a parameter pair (u,v). So each u and v parameter pair can be evaluated to a 3-dimensional point. You are evaluating the derivative at the parameter space but the parameter space of the surface is mapped to the 3-dimensional Euclidean space. Eventually, you will get a 3-dimensional derivative.

Mitch981521, 2019-03-16:
Thank you very much for the detailed answer.  Yes, I was thinking that the function was supposed to return the derivative on the parametric space on the surface, not euclidean space.  I also did not realize P is 3-dimensional, so that's why I was thinking everything was float.  I should've thought about gradient....I don't know why I didn't see that. 

What I am needing is dz/dy (trajectory tangent) as I travel in the euclidean y direction, because the surface tangents need to correspond to the trajectory of our robot.  So this means that for my purpose, the parameter space should ideally align with the euclidean space, right?  (u direction should be aligned with x, v should be aligned with y).  Otherwise, if I take derivative w.r.t. to v (or u), the dx, dy, dz won't mean anything unless I am traveling in the v (or u) direction.  I just want to be sure I'm visualizing this correctly.  Thank you again

orbingol, 2019-03-16:
This issue looks resolved. Closing it.

closed 2019-03-16 by orbingol