orbingol/NURBS-Python #48: Make visualization modules to cache figure data

orbingol, 2019-03-23:
Visualization modules in NURBS-Python were never intended to be a part of the core library. In the beginning, they were added to visually debug the evaluation results in a faster way. Therefore, they weren't designed to be efficient but just visualize whatever sent to them from the `render` method. When needed, they received some upgrades and new features and these are are mostly connected to the core library updates.

As of geomdl v5.2, it is very easy to say that all visualization modules are now a part of the core library but due to the preliminary idea in their design, they seem to be now a little bit outdated and more importantly, they work slow. Therefore, they need to be improved.

This issue is opened due to some problems encountered while implementing the feature request #46 and it would be a good idea to revise the visualization system a little bit without changing its API. As a result, this issue extends #46.

The following changes are proposed for now:

1. At every call, `render` method resets the plot cache in the visualization module and transfers the point data again to the visualization module. The visualization module converts all these into its internal data structure and as a result, visualization gets slow. There can be a conditional check in the `render` method of the geometry class to use existing plot data in the vis module if there are no changes in the geometry data. In addition, it is always good to have an optional argument, such as `force`, to force re-transfer of the point data to the vis module.
2. The visualization modules can cache the figure data for faster viewing. This cache should be connected to the changes and the `force` parameter in the `render` method of the geometry class.
3. Add key press events to change visibility of objects, if possible

Tcll, 2019-04-17:
Hey if you need help here, I can provide some meta-performance tips to help improve your general python code :)

eg:
- use `__slots__` to reduce a class's memory footprint/consumption.
(this is not ideal for small situations with under 100 total (all classes combined) instances)

- avoid the dot operator when inside a loop where possible (`__gtattr__`/`__getattribute__` is slow).
(in some cases like int and float, local referencing from C object methods can actually be slower)

- avoid `isinstance()` unless absolutely necessary, and use direct `type()` or `instance.__class__` if safe.
(while faster than `type()`, your code can be fooled when using `instance.__class__` where `__class__` is overridden on creation)

let me know if you need more :)

orbingol, 2019-04-17:
@Tcll thank you so much for your suggestions. I am aware of all these facts and their pros and cons. Some of them are there for simplification and documentation purposes. Some of them are there because I was asked to. Some of them are there for fixing the compatibility issues and make some algorithms compatible. Before geomdl v3.0, I also had to consider backward compatibility issues with an another library. Not all of them are best practices but that's why this project is open source :) You can always contribute by means of pull requests to make it better ;)

Just for my 2 cents; for pure Python libraries only using Python Standard Library, you can always get better performance with PyPy. You can check the Travis CI logs for comparison between CPython and PyPy. It is even possible to get better performance with Numba.

Regarding to `__slots__`, I have experienced some problems when trying to provide Python 2 and 3 support together with `six`, use `ABC`s and property protocol at the same time. I tried to find a solution that time but currently, I don't even remember what the problem was. If you could help me on that, I would be glad.

I am currently in need of a high performance visualization library and this is basically the main topic of this issue. Performance improvements for VTK also work but I am pretty sure that whatever they recommend on their documentation and email lists don't work.

orbingol, 2019-04-18:
> Regarding to `__slots__`, I have experienced some problems when trying to provide Python 2 and 3 support together with six, use ABCs and property protocol at the same time. I tried to find a [...]

I wasn't going to add more comments but just for curiosity, I have checked the code again. When I had the issues (those that I don't remember, it has been a really long time), I was using `six.with_metaclass` function for `ABC` compatibility. I have replaced it with `six.add_metaclass` decorator which works way better than `with_metaclass`. I think I can go back and implement the `__slots__` logic again. I might even have the patch in one my local branches :)

I also would like to add a note on `isinstance`. I have added instance checks for convenience but then I noticed that I can replace most of them with the parametric dimension checks and I think this is the best way due to the nature of instance tracking in Python. So, with geomdl 5.0 release, I added `pdimension` property for representing the parametric dimension of the spline geometry. I have been converting `isinstance`s since then, when I find time and energy to do the changes :)

I only have doubts on how to remove list and tuple instance checks without breaking Python 2 and 3 compatibility.

Let's answer the others too :) I am using Python properties for convenience, mostly check (or fix) the user input. If you are ready for solving the issues on your own, you can use the "underscore" variables directly. I see some users of geomdl around me are using those variables directly.  `__all__` for controlling `from geomdl import *` guys (I know this is a bad practice but since it works, you cannot prevent people to use it) and it also helps to mask some internal functions in the documentation. I am losing track if I do those changes inside Sphinx  `automodule` arguments. I took the idea of `@export` decorator and usage of `__all__` in every module (i.e. Python file) from one of the well-known Python libraries, I need to take a look at my notes to remember that library name. `multiprocessing` is there for experimentation purposes and they are not forced by default. For my case, it speeds up tessellation and voxelization operations and I use the results to compare with my yet-unreleased PhD research :). You can definitely use MPI or MP interfaces (or other acceleration libraries/methods), but they have their own problems, overheads and implementation problems.

Tcll, 2019-04-18:
Ah I see, I guess that makes you the first highly experienced pythonist I've run into since 2015 or so :D

honestly, I would contribute, but as you can probably guess, I'm into quite a few of my own large projects already, practically swimming on my plate... lol
but I don't mind providing advice and maybe a few small improvement suggestions here and there :)

and yes, I do know about alternatives like Pypy and Numba and such (I really wish I could use Pypy, but it's not portable and doesn't fully support windows, linux, and mac on x86 and x64)
I'm currently getting away with Anaconda stripped down to around 12MB per interpreter for my projects (working on the bare minimum of what I can)

for your `__slots__` issue, I might be able to help, but I don't really understand the purpose of `abc` (which is also a nightmare module I'm trying to get rid of), and haven't really used `six`

for a high perf visualization library, personally I'd probably just build what I can myself on top of modern (at least using vector lists) OpenGL...
but that's likely way out of reach from the current scope of this project, so for a substitute, I remember Ian Mallet (aka geometrian) commonly using matplotlib for cases like this, which I commonly hear is quite performative, but by how much, I'm not sure...
I hear the same thing about numpy, and I think we both know how just performative that is if we both actively avoid using it :D

I don't have much experience with `vtk` honestly, but if it has anything to do with `tkinter` I can already presume it's going to be slow :P

and with your updated comment, you actually opened my eyes, as never once have I put 2 and 2 together to think that `__all__` would control `from name import *`
I should actually start doing that myself now that I look at it, thanks for that! :)

and yeah true, any dependencies you use adds some major bulk, as well as opens your source up to deprecation issues from them.

anyways, seeing as the support for specific modules like `abc` and `six` with properties and such was likely requested, I don't think there's too much I can do to help with performance, but I'll have a look and see if I can spot anything that might be able to be done better.
I wasn't sure at the start, but you seem to have this pretty under control for what you can do :)

also, to avoid turning this into a long-winded conversation you can find me pretty easily by my tag "Tcll5850"
I tend to hang around on forums and such, but my gmail works too if you'd like to hold a discussion. ;)
(would be nice if GH had the option to hold a private discussion were we could better talk about this)

orbingol, 2019-04-18:
All comments, suggestions, bug reports, "I just wanted to say hi", short (but not too short), long, super long messages are always welcomed :) I am good with GH issues as long as people don't mind posting publicly, My email is also open for any kind of discussions (long or short doesn't matter): contact@onurbingol.net

Actually, PyPy 32bit version for Windows is available for download. I have tried it a couple of times and it seemed to work fine. Even some CPython modules install and work fine (you can check geomdl Circle CI logs for PyPy-based docker image building). But I use the Linux version of PyPy for performance testing of the algorithms.

I have also noticed a small performance difference between the default distro (the one you download from Python.org) and the Anaconda. In fact, Anaconda runs slower than the default Python distro (it could be anything, just I kept the module list exactly the same for testing). It has been a while since I did the last comparison test but this is what I remember.

You may want to frequently check [the unofficial packages directory](https://www.lfd.uci.edu/~gohlke/pythonlibs/) and be ready to get your hands dirty with compilers, if you prefer to use the Python,org distro instead of Anaconda.

ebo, 2020-05-30:
has anyone tried adding numba support?

orbingol, 2020-05-30:
This issue has been fixed in v6.x branch

closed 2020-05-30 by orbingol