orbingol/NURBS-Python #49: Normal vectors to curves are incorrect.

carthurs, 2019-06-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Configuration:**
 - OS: [e.g. Windows 10, Ubuntu 18.10]
 - Python distribution: [e.g. OS default, python.org, Anaconda]
 - Python version:
 - geomdl install source: GitHub / PyPI / Anaconda
 - geomdl version/branch:

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
Add any other context about the problem here.

orbingol, 2019-06-04:
Would you also check if it occurs any other place and make a PR against `devmaster` branch?

carthurs, 2019-06-04:
PR #50 made - I think there's still work to do, but it fixes that one case and should not impact other cases (i.e. 3D curves). Probably need a fix in 3D too.

commit 73119bc6 (2019-07-11): Curve normal eval fix, closes #49 and closes #50
closed 2019-07-11 by orbingol