orbingol/NURBS-Python #51: Getters and setters related to control points should use the new Manager classes

orbingol, 2019-06-05:
Currently, `ctrlpts`, `ctrlptsw` and `weights` properties accept a list of points (or values) and the order of the points are described in the documentation. Addition of B-Spline and NURBS volume geometry type makes setting of the control points a little bit confusing. The users do not need to know the internal structure of the library and it also serves to the purpose of geomdl: keeping the things as simple as possible.

As a result, it would be a good idea to replace the get-set values of the corresponding properties with a point manager class under `control_points` module. This is going to result in an API change starting from v5.3.0 but it will make control points setting less confusing, especially for `Surface` and `Volume` users.

cshorler, 2019-10-01:
What I've just looked at using NotifyList for ctrlpts could offer another potential approach - pseudo code

```python
class CPList(NofityList):
  pass

class SomeVolume(...):
  def __init__(self, ...):
    self._control_points = (CPList(), CPList(), CPList())
  @property
  def ctrlpts(self):
    return self._control_points
```

The tuple replaces the individual accessors, unpacking as for any other tuple.  Some additional protection `CPList` would give a more specific `TypeError` if getting your indexing wrong to a coordinate level 

Subclassing tuple would be a easy way to provide a smart object wrapper directly on the returned value from ctrlpts, if you want to refer to something as `obj.ctrlpts.u` rather than `obj.ctrlpts[0]`  Although, I'd think `u, v, w = obj.ctrlpts` is more direct.

orbingol, 2019-10-02:
I am not sure what you mean by NotifyList but what the container needed to store control points is something like `list<float>`. Direct subclassing of Python's `list` is possible but then the majority of the methods is needed to be overridden to make sure the data structure and the type is valid.

For instance, that specific subclass needs to store some additional fields, such as spatial dimension and number of control points in all parametric dimensions (as a tuple) which simple determine the "jumping distance" between the parametric dimensions. As a result, all data can be stored as a list of floats but whenever it enters a for loop, then additional internal processing will be required to determine the control point returned. As a matter of fact, Python's `array` class might fit better in this case; `from array import array`.

From what I understand, your suggestion seems to require storing the same control points in different lists but different ordering. I believe that can be achieved by simple caching (for instance, in NURBS module, control points and weights are cached in a `dict` since the object stores "weighted control points" by default).

orbingol, 2020-07-07:
This has been implemented in `6.x-dev` branch,

closed 2020-07-07 by orbingol