orbingol/NURBS-Python #53: `ZeroDivisionError` on large bézier knotvectors

jovobe, 2019-06-06:
**Describe the bug**
A `ZeroDivisionError` occurs with the message "float division by zero".

**To Reproduce**
Steps to reproduce the behavior:
Run the following example:
```python
from geomdl import BSpline
from geomdl import utilities
from geomdl.visualization import VisMPL

curve = BSpline.Curve()
curve.degree = 3
curve.ctrlpts = [[0, 0], [1, 1], [2, 0], [3, 1], [4, 0], [5, 1]]
curve.knotvector = [0, 0, 0, 0, 0, 1, 1, 1, 1, 1]
curve.delta = 0.01
curve.vis = VisMPL.VisCurve2D()
curve.render()
```
**Expected Behavior**
This should yield a Bézier curve, right?

**Configuration:**
 - OS: macOS 10.14.5
 - Python distribution: brew
 - Python version: 3.7.3 64-bit
 - geomdl install source: PyPI
 - geomdl version/branch: 5.2.7

**Additional Details (Optional)**
The code above works, if we only define three ctrlpts and a knotvector of length 6:
```python
curve.ctrlpts = [[0, 0], [1, 1], [2, 0]]
curve.knotvector = [0, 0, 0, 1, 1, 1]
```

orbingol, 2019-06-06:
The knot vector given is for `degree = 4` because number of zeros and ones should be `degree + 1` to generate a Bezier curve. If `degree = 4`, then using the equation on p.82 of The NURBS Book `m = n +p + 1`, we can see that input number of control points is 1 bigger than the required one. On the case of `degree = 4` and a Bezier curve, removing 1 control point should work fine. Otherwise, set `degree = 5` and update the knot vector.

For Bezier curves, the computation is always very simple: if `degree = k`, then number of control points `k + 1` and knot vector size `k + 2`, which also can be proved by the equation `m = n + p + 1`

On the other hand, the library should check this problem and inform the user. That's a different kind of problem and it should be checked.

orbingol, 2019-06-06:
Well, actually there is nothing to do for the `m = n + p + 1` checking in the library. It is working correctly but your unique combination of degree, number of control points and size of knot vector (even they are wrong) just fits to the equation:  m = n + p + 1 => 10 = 3 + 6 + 1

jovobe, 2019-06-06:
Okay than, even if it does not yield a Bézier curve the example should work, right? Because it is still a valid B-Spline curve.

orbingol, 2019-06-06:
I don't think so.

jovobe, 2019-06-06:
> I don't think so.

Why not? All rules are respected. It is a simple three degree B-Spline curve with six ctrlpts and a decreasing knotvector. This is absolutely legal if you follow Piegls book.

orbingol, 2019-06-06:
What you are missing is the equality `m = n +p + 1` is only one of the long set of rules. As you said, the stability of the B-Spline evaluation depends on all the rules to be satisfied. Satisfying only one rule is not enough.

jovobe, 2019-06-06:
> Satisfying only one rule is not enough.

Could you be a little more specific? Which assumption is not fulfilled?

orbingol, 2019-06-07:
I am guessing you want some academic reference. Usually, this is the part where all the libraries and books "leave the proof to the reader" because it can get a little bit messy to explain. I remember reading about this situation related to the concept of knot multiplicity in some pre-1990 paper (or a book). Unfortunately, I don't have access to these references right now. On the other hand, not all research papers have up-to-date correct information as some concepts could have invalidated/changed through the years.

If you feel geomdl has some incorrect assumptions or computations, please don't hesitate to open a PR with the fix. All contributions are welcomed!

closed 2019-06-07 by orbingol