orbingol/NURBS-Python #57: ray.intersect tolerance is unrealistic

cshorler, 2019-08-23:
**Describe the bug**
The default tolerance (tol=) of the ray.intersect is set smaller than the machine epsilon.  Leading to RayIntersection.SKEW results (sometimes incorrectly)

**To Reproduce**
test case in my fork branch and workaround:
https://github.com/cshorler/NURBS-Python/tree/relax_ray_intersect_tolerance
(need to rebase this on master if PR is required before integrating other PR - happy to do either).

**Expected Behavior**
My quick fix was to change the default tolerance to be a multiple of machine epsilon after investigating a few likely causes in linalg module.  In the long term, I think the correct thing to do would be change the implementation to improve the accuracy.

**Configuration:**
 - OS: Tested on Windows 10 (Python 2.7), Linux openSUSE (Python 3.6)
 - Python distribution:. Anaconda & OS default
 - Python version: see above
 - geomdl install source: GitHub
 - geomdl version/branch: master, and others

orbingol, 2019-08-24:
Thanks @cshorler but I couldn't see any commit related to change in the tolerance value. Would you please create a PR against `devmaster` branch? I might be able to merge a small change in the next minor release.

cshorler, 2019-08-24:
Okay, will do

orbingol, 2019-08-25:
PR #59 is merged as a temporary fix for geomdl v5.x. PR #60 will be considered for the next major release.

Thank you very much for your contributions @cshorler. I will add your name and GitHub handle to the contributors list.

PR #60 (2019-10-07): use augmented arithmetic to ensure adequate precision for calculating…
… ray intersections

fixes #57
closed 2020-08-08 by orbingol