orbingol/NURBS-Python #63: a line seems useless

silentmissile, 2019-09-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Configuration:**
 - OS: [e.g. Windows 10, Ubuntu 18.10]
 - Python distribution: [e.g. OS default, python.org, Anaconda]
 - Python version:
 - geomdl install source: GitHub / PyPI / Anaconda
 - geomdl version/branch: 5.9.2

**Screenshots (Optional)**
If applicable, add screenshots to help explain your problem.

**Additional Details (Optional)**
in fittings.py function compute_params_curve, when calculate the chord length, at line 441, the last element of cds is assigned -1

cds is a list of chord length, it shouldn't be assigned to a fixed value, if needed, the first element of cds can be assigned 0, other elements should not be a fixed value

and it's length should be equal to points length, not plus 1 (in line 440)

after some review of the literature book and the code, I think it's useless. Maybe you developer have some extra idea, I'll be very happy to see if you provide detailed introduction of your idea.

orbingol, 2019-09-16:
@silentmissile I couldn't understand whether this is causing a computational problem or not.

silentmissile, 2019-09-17:
this line won't cause any mistake in computation, it just occupies some more memory and a little more computation load

if you remove line 441 and reduce the length of cds equal to points length, the code would slightly accelerate a little speed

by the way, if anybody else wants to develop base on your code, for example algorithm A9.6 in section 9.4.2 in the reference book, my proposal would make your code exactly the same as the reference book, other developers can understand more clearly. or it would cause confusing.

orbingol, 2019-09-17:
> by the way, if anybody else wants to develop base on your code, for example algorithm A9.6 in section 9.4.2 in the reference book, my proposal would make your code exactly the same as the reference book, other developers can understand more clearly. or it would cause confusing.

I was thinking that I have already achieved that. I don't even change the variable names when I implement the algorithms :) But, there is always a difference between an algorithm and an implementation. For instance, using Python's list comprehension, it is possible to code a single line which refers to a small-medium for loop. 

Another thing is that I don't think there is an implementation of Algorithm A9.6 in geomdl. That algorithm is a weighted and constrained LS approximation. The documentation of `approximate_curve` refers to the Section 9.4.1 which is possibly the equations (it has been some time since I have implemented those functions).  I don't remember why I did it like this. If the algorithms are implemented, then the documentation should refer to the specific algorithm. Otherwise, it refers to the equation or the section. I guess this is the part which is not clear.

I guess I might add Algorithm A9.6 to the list of implementations before the release of v5.3.0

orbingol, 2019-09-18:
I had a chance to take a deep look at what you suggested. You are definitely right, `compute_params_curve` needs some optimization but according to my tests, what you proposed actually changes the stored values in the array and affects the fitting results.

`fitting` module also needs some improvements but there won't be any in geomdl 5.x. I will work on those for geomdl 6.x. Contributions are always welcome!

closed 2019-09-18 by orbingol