orbingol/NURBS-Python #65: NURBS is not correct when applied a weight vector. if the weight vector is not all 1.

DeathGun007, 2019-09-26:
from geomdl import NURBS
from geomdl import compatibility as compat

crv = NURBS.Curve()
p_ctrlpts = [[1, 1], [2, 1], [3, 0], [4, 4], [3, 9], [10, 2]]
p_weight = [1, 1, 1, 1, 3, 1]

t_ctrlptsw = compat.combine_ctrlpts_weights(p_ctrlpts, p_weight)

after, Weight = compat.separate_ctrlpts_weights(t_ctrlptsw)

crv.degree = 2
crv.ctrlpts = t_ctrlptsw
crv.knotvector = [0, 0, 0, 1, 2.5, 3, 5, 5, 5]
crv.delta = 0.005
points = crv.evalpts

from geomdl.visualization import VisMPL
crv.vis = VisMPL.VisCurve2D()
crv.render()
#the result was the coordinates change of the control points. But I just change the weight, why?

DeathGun007, 2019-09-26:
Hope I didn't make any mistake. I just found the coordinates changed with the weight vector. That shouldn't happen, right?  Hope you can give me some help. thank you !

orbingol, 2019-09-26:
@DeathGun007 I couldn't understand why you are using the following code?

```
t_ctrlptsw = compat.combine_ctrlpts_weights(p_ctrlpts, p_weight)

after, Weight = compat.separate_ctrlpts_weights(t_ctrlptsw)
```

It feels completely unnecessary as your points and weights are already in separate lists.

The following code will be enough for the example given:

```
p_ctrlpts = [[1, 1], [2, 1], [3, 0], [4, 4], [3, 9], [10, 2]]
p_weight = [1, 1, 1, 1, 3, 1]

crv = NURBS.Curve()
crv.degree = 2
crv.ctrlpts = p_ctrlpts
crv.weights = p_weight
crv.knotvector = [0, 0, 0, 1, 2.5, 3, 5, 5, 5]
crv.delta = 0.005
points = crv.evalpts

from geomdl.visualization import VisMPL
crv.vis = VisMPL.VisCurve2D()
crv.render() 
```

assuming that `p_ctrlpts` is a list of `(x, y)` but not `(x * w), (y * w)` where `w` is the weight associated with the control point.

There is also a wiki page which explains the details: https://github.com/orbingol/NURBS-Python/wiki/Control-points-getters-and-setters

DeathGun007, 2019-09-27:
Thank you so much! You really help me out.

orbingol, 2019-09-27:
@DeathGun007 you are welcome! It seems that everything is clear now and I am closing this issue. Please feel free to open a new issue if you have more questions.

closed 2019-09-27 by orbingol