orbingol/NURBS-Python #66: Sphere import from rhino

akfhr, 2019-09-27:
Im new to geomdl, facing some problems with importing primitives like e.g. a sphere. 

The sphere is displayed in Rhino correctly. 
However, after converting the 3dm file with on2sjon, and importing again with geomdl, the gemeotry is significantly different from what I would expect:

import os
from geomdl import NURBS
from geomdl import exchange
from geomdl.visualization import VisMPL as vis

# Import surfaces
surfs = exchange.import_json("sph_r_0p5m.json")

# Plot the control point grid and the evaluated surface
surfs[0].sample_size = 100
surfs[0].delta=0.05
surfs[0].vis = vis.VisSurface()
surfs[0].render()

[geometry.zip](https://github.com/orbingol/NURBS-Python/files/3662286/geometry.zip)

Im grateful for any help, thanks! 
[image]
[image]

orbingol, 2019-09-27:
The problem is most probably related to how geometry is exported from the Rhino file but not geomdl itself. Would you please open an issue in `rw3dm` project?

akfhr, 2019-09-27:
Dear orbingol, 
I've opend an issue there. Thanks!

closed 2019-09-27 by orbingol