orbingol/NURBS-Python #67: NURBS derivatives

cshorler, 2019-09-29:
I made a test of NURBS derivatives using a circle, I don't believe the outputs to be correct (see picture).  I also note that NURBS derivatives are documented in more recent papers than the NURBS book - so perhaps this isn't expected to work.

I'll provide a test for this at the next opportunity

Manually (Tested on Linux & Windows, Python 2 & 3):

1. Create a 7pt NURBS circle (also not convinced the 7pt construction in geomdl.shapes is correct after my back of the envelope calculation for a unit circle - it looks isosceles rather than equilateral?).  Also tested with 9pt (see picture)

2. Create derivatives

3. Plot the results

I was expecting circles.
[image]

orbingol, 2019-09-30:
I think there is a problem with the weights (or weight assignments corresponding to the control points). Would you please provide the code that generates this figure?

orbingol, 2019-09-30:
@cshorler I just forgot to mention the Evaluator interface ([ref](https://nurbs-python.readthedocs.io/en/latest/module_evaluators.html)). It allows using different algorithms for computing points and derivatives with minimal changes to the base classes. I'd suggest implementing a new evaluator class for any new/updated algorithms.

cshorler, 2019-10-01:
The rendering in matplotlib looks okay to me, it's the control point coordinates that look wrong to me - the triangle sides are not of equal length.

w.r.t. Derivatives it would be good for me to better understand why this derivatives issues occurs ... so I will try for a deeper understanding and explore alternate evaluator implementations.

For reference here's similar code to what was used to generate the picture - it seems I didn't save / stash the original!  Do you want me to attempt to remove the numpy dependency from VisMPL?

```python
import math
import matplotlib.pyplot as plt
from geomdl import NURBS
from geomdl.linalg import linspace
r = 1.
a, h = 3. * r / math.sqrt(3.), 1.5 * r
crv = NURBS.Curve()
crv.degree = 2
crv.ctrlpts = [(0., -r), (-a,-r), (-a/2,-r+h), (0., 2*h-r), (a/2, -r+h), (a, -r), (0., -r)]
crv.knotvector = [0.,0.,0., 1./3, 1./3, 2./3, 2./3, 1.,1.,1.]
crv.weights[1::2] = [0.5, 0.5, 0.5]
derpts = [crv.derivatives(u, order=1) for u in linspace(0., 1., 50)]
legend_proxy, legend_names = [], []
fig = plt.figure(figsize=(10, 8), dpi=96)
ax = fig.gca()

cpplot, = plt.plot(*zip(*crv.ctrlpts), color='blue', linestyle='-.', marker='o')
legend_proxy.append(cpplot)
legend_names.append('control points')

curveplt, = plt.plot(*zip(*(t[0] for t in derpts)), color='black', linestyle='-')
legend_proxy.append(curveplt)
legend_names.append('curve')

derplt, = plt.plot(*zip(*(t[1] for t in derpts)), color='green', linestyle='-', marker='+')
legend_proxy.append(derplt)
legend_names.append('1st derivative')

plt.legend(legend_proxy, legend_names)
ax.set_aspect('equal')
ax.set_xlabel('x')
ax.set_ylabel('y')
plt.show()
```

orbingol, 2019-10-02:
Thank you for sharing the code @cshorler. I'll test it in a couple of days.

> Do you want me to attempt to remove the numpy dependency from VisMPL?

I tried that once but it seems like an impossible job :) Pretty much all the visualization libraries that I used in Python requires Numpy in some way. No need to worry about that.

Just a note for myself: I need move `vis.py` to `visualization` package to fix compatibility/dependency issues with Cython-compiled version of geomdl.

ted-dgk, 2020-01-15:
Just a comment guys as I am poking around the nurbs landscape and I noticed this post with data.

I tried the data from your example in 3 other systems:

[http://nurbscalculator.in](url) and tinynurbs (C++ header only) and they both give a Wankel engine kind of 3 sided rotor - that is if I am doing it right.

Also tried in verb and it gives your circle. The verb test was done in javascript.

orbingol, 2020-01-16:
@ted-dgk thank you so much for the tests! It has been real busy for the last couple of months, I couldn't find time to test this issue thoroughly.

There was another issue (I believe it was #35) about computing the derivatives of a NURBS circle. I remember checking the exact same thing with the [NURBS toolbox](https://octave.sourceforge.io/nurbs/) while I was trying to find a fix. I think there were some notes on the source code of the NURBS toolbox.

cshorler, 2020-01-16:
I've read both of the more recent NURBS derivatives papers now.  Understood
some of it, still working on the rest.  I'm not sure of exact understanding
yet.  We should implement both in my opinion.  I'm struggling to find time,
but I will at some point as I still need this for more general NURBS (the
circle is easy to spot the error, but not a real world example)

Also note very first point in this issue about the circle construction - I
believe this is also a bug.

orbingol, 2020-01-16:
> Also note very first point in this issue about the circle construction - I believe this is also a bug.

@cshorler, you mean the geomdl.shapes issue?

cshorler, 2020-01-17:
Yes

orbingol, 2020-01-17:
@cshorler thanks. Let me open an issue on the `geomdl-shapes` issue tracker on that

cshorler, 2020-06-18:
@orbingol 

I started to explore the 6.x-dev branch a little bit, some interesting changes / explorations.

some initial comments on files I'm also likely to submit PR on where I might revert locally before making PR.

* geomdl/base.py 
- I guess the introduction of GeomdlFloat is a test to manage fp roundoff issues, I'm not sure the way this is implemented is a good design decision w.r.t. fp accuracy vs. needs.

* geomdl/abstract.py - 
- the SplineGeometry degree property was changed in type for curves, probably this needs to be abstract at that level and concrete at a lower level in the inheritance hierarchy so it can be basic type for curve and list type for surfaces and others.

* geomdl/control_points.py -
- CPManager seems to be used more widely.  If the aim is to associate data to control points - some of the changes I'll propose for this issue might be interesting as they will simplify how this can be implemented (I think - use case not yet completely understood).

orbingol, 2020-06-18:
6.x-dev is still under development, some new features are not completed in the way I indented and it is not ready for a release. I am thinking of an alpha release at the end of this month and a roadmap for the future development.

Comments are always welcome but it might be too soon for PRs.

cshorler, 2020-06-18:
I assumed this, okay will do

orbingol, 2020-06-18:
> I assumed this, okay will do

@cshorler thanks!

closed 2020-08-08 by orbingol