orbingol/NURBS-Python #68: Slicing

cshorler, 2019-09-29:
I have a piece of code which calculates control points for an arbitrary arc using vector calculus.  One possible way to achieve outputs is to pre-calculate nice fraction angular sector weights (e.g.  1/2).  Only calculating the final weight explicitly, possibly this isn't a good approach for distributing the error - but what's notable is slicing is not supported, so you have to workaround this - making the code slightly less concise than it could be.

The example is for weights, but I imagine the problem / feature need is evident on all attribute lists of curves and surfaces.

I've started working on a possible way to handle this the branch below:
https://github.com/cshorler/NURBS-Python/tree/slicing_and_nurbs_derivatives

cshorler, 2019-10-01:
now successfully passes the test I wrote

orbingol, 2019-10-07:
Thank you @cshorler. I think slicing feature will be a nice addition to geomdl.

cshorler, 2019-10-12:
I'll submit a PR for v6

orbingol, 2019-10-13:
`6.x-dev` is very unstable at the moment. I am doing some major refactoring to simplify some algorithms which will improve the performance in general. I will post a list of current and future changes for `6.x` in a couple of days. so you will be able to track the details.

cshorler, 2019-10-13:
seems to work on 6.x-dev, see PR opened.  Also, the NofityList mentioned in the comment on your other enhancement is used in the implementation.

orbingol, 2019-10-13:
Would it be possible to create a separate PR for the NotifyList implementation?

cshorler, 2019-10-15:
Will do!