orbingol/NURBS-Python #70: How to save Surface as CAD Support file IGS/STEP ?

selvakarna, 2019-10-15:
How to save Surface as CAD Support file IGS/STEP ?

orbingol, 2019-10-16:
Exporting to IGES and STEP feature is still work-in-progress.

selvakarna, 2019-10-16:
Can you share sample of step export from surface or mesh ? 

algorithms?

silentmissile, 2019-10-17:
you can try to use FreeCAD

https://anaconda.org/search?q=freecad
the package follows newest Anaconda

here is some example code, you can refer to it
https://forum.freecadweb.org/viewtopic.php?f=22&t=38218&p=325619&sid=903589966cb65fc527f65ce3c1812b1a#p325619

or you can try pythonocc
https://github.com/tpaviot/pythonocc-core

it has some recent comments, but the last formal release was 2 years ago, and I tried to install it by the latest Anaconda, but failed. By my test only Anaconda 5.2 is compatible with it

orbingol, 2019-10-17:
I am guessing that this issue is somewhat resolved. Hopefully, I will also release an option for exporting/importing IGES and STEP files very soon :)

closed 2019-10-17 by orbingol