orbingol/NURBS-Python #73: Quadratic surfaces get rendered as cubics

virtualritz, 2019-12-03:
I'm trying to render [a JSON file](https://github.com/orbingol/NURBS-Python/files/3919237/sphere.json.zip).

This is a unit radius NURBS sphere generated in Rhinoceros and converted from OpenNURBS to JSDON via `on2json`.

Importing & exporting this as a JSON again with NURBS-Python yields a perfect sphere. So the data seems to be handled correctly.

However, when visualized/rendered via `geomdl.visualization` it looks like a cubic surface would using these CVs, not like a sphere.
[image]

virtualritz, 2019-12-03:
On that note: fixing this may present a good opportunity to ensure that higher order surfaces – quadratics, quintics, hexics, etc. – get visualized correctly.

orbingol, 2019-12-04:
Have you tried to change surface evaluation delta to something smaller, e.g. 0.01?

virtualritz, 2019-12-18:
If I set the delta to this value the script never even opens the window. It seems to be hanging. I stopped the process after 5mins.

Output:
```
Traceback (most recent call last):
  File "trim_GIS.py", line 68, in <module>
    sphere.render(colormap=cm.copper)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/abstract.py", line 1949, in render
    self.tessellate(force=force_tsl)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/abstract.py", line 2005, in tessellate
    trims=self.trims, **kwargs)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/tessellate.py", line 162, in tessellate
    tessellate_args=self.arguments, **kwargs)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/_tessellate.py", line 135, in make_triangle_mesh
    vlst, tlst = tsl_func(vertex1, vertex2, vertex3, vertex4, vrt_idx, tri_idx, trim_curves, tsl_args)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/_tessellate.py", line 324, in surface_trim_tessellate
    t1, t2, status = ray.intersect(edges[idx2], trim_ray)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/ray.py", line 147, in intersect
    return _intersect2d(ray1, ray2, tol)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/ray.py", line 166, in _intersect2d
    return _intersect3d(ray1_3d, ray2_3d, tol)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/ray.py", line 198, in _intersect3d
    if linalg.point_distance(ray1_pt, ray2_pt) < tol:
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/linalg.py", line 321, in point_distance
    dist_vector = vector_generate(pt1, pt2, normalize=False)
  File "/Users/moritz/Library/Python/3.7/lib/python/site-packages/geomdl/linalg.py", line 184, in vector_generate
    ret_vec.append(ep - sp)
KeyboardInterrupt
```

virtualritz, 2019-12-18:
Ok, I let it run and after 6mins I get this image – same bug, just denser tessellation. This does still look like a quadratic being rendered as a cubic. 
[image]

orbingol, 2019-12-19:
I am sorry, I have completely missed the part that you have converted the unit sphere using `rw3dm`. There was another bug report on the `rw3dm` issue tracker and it was resolved temporarily. Please refer to the that solution for the moment: https://github.com/orbingol/rw3dm/issues/8

I am closing this issue, it is not related to `geomdl` but `rw3dm`.

closed 2019-12-19 by orbingol