orbingol/NURBS-Python #74: How to grade the NURBS curve with the grading point when use CAD?

friends110110, 2019-12-23:
When provide some knot points on the NURBS curve in CAD , how to grade the NURBS curve with the grading point?

Is there java NURBS lib for me ?

I search for some research: we parse the point from DXF(CAD file format), the weight associate with the NURBS curve in DXF is default 1, then interpolate the knot points and get the control points, then could draw the NURBS curve , using python lib python-nurbs lib

# when deal with polyline in dxf
from geomdl.fitting import interpolate_curve
....
degree = min(pointSize - 1, 3)
curve = interpolate_curve(points, degree)   # points which on the NURBS curve
vertices = curve.evalpts   # this is knot points ,and we can get control points

orbingol, 2019-12-23:
@friends110110 I couldn't understand what you mean by "grading point". However, `fitting.interpolate_curve` would give you a `Curve` object. It is possible to compute or get pretty much everything when you have the `Curve` object. I would suggest taking a look at the `Curve` class documentation for more details.

friends110110, 2019-12-24:
@orbingol I update the description. The grading point is what scale the knot points .
Firstly, I provide some knots points in the nurbs curve , so we can infer the argument on the nurbs curve.
Secondly, I provide the grading rule table whch could scale the specify knot points(not all poins), How to scale the knot points and draw another NURBS curve (shape？)like the step firstly ?

orbingol, 2019-12-24:
@friends110110 the curve points (that are what you name "knot points" and I generalize them as "evaluated points" and they are called "data points" in the `fitting` module) are the outcome of the NURBS functions. `Curve` instances store them for caching purposes. Interpolation and approximation functions are designed to find a valid knot vector and the control points (depending on the input degree) then re-evaluates the geometry using the NURBS functions (that is the default behavior of the `Curve` class). Therefore, a different set of "data points" (generated using your table) would result into a different `Curve` object.

You can use the container classes in `multi` to visualize multiple geometries.

orbingol, 2019-12-31:
@friends110110 I am closing this issue now. Please reopen this issue if you have any other questions.

closed 2019-12-31 by orbingol