orbingol/NURBS-Python #75: T-spline visualization

OVGULIU, 2020-01-23:
Hello @orbingol ,

Good day to you!

In the next version of NUBRS-Python can we import(/export/...) **T-Spline** from Rhino@?

Best regards

orbingol, 2020-01-23:
@OVGULIU you mean T-spline support for NURBS-Python, I presume. It is not a priority for now but it shouldn't be that hard after the release of geomdl v6.x.

AAAWWWAA, 2020-01-23:
@orbingol , Nice to hear that! Thanks!

GarthSnyder, 2021-02-03:
I was looking at some T-Spline issues related to Fusion 360, and it seems the T-Spline file format is undocumented at this point. There used to be a spec you could download from T-Splines when it was a separate company, but since the Autodesk acquisition their web site is no longer maintained. The Internet archive does not have a copy of the actual file, and nothing useful turns up on Google.

The format is text-based (or at least, there is a text-based version), so it's potentially reverse-engineerable. But I suspect it's not trivial.