orbingol/NURBS-Python #76: Possibility to export / import using arbitrary file-like object

portnov, 2020-02-13:
Currently `exchange.import_json`, `exchange.export_json` and friends can work only with file name; but as far as I can see from code, there is nothing preventing them to work with arbitrary file-like objects, StringIO for example.

**Describe the solution you'd like**
IMHO existing methods could just test if `file` argument is `isinstance(str)`, then treat it as file name, otherwise treat it as a file-like object. Another solution is to add new methods like `exchange.export_json_???()` (can't think of good name right now), that would work with file-like objects, and leave existing methods as they are.

**Additional context**
I'm going to use these methods from Blender's addon, so I'd like to use Blender's text buffers instead of actual on-disk files.

orbingol, 2020-02-15:
@portnov thank you for the suggestion.

geomdl v5.x has similar functions for different exchange formats, such as OBJ. For instance, `exchange.export_obj_str` returns a string and `exchange.export_obj` saves the string as a file. I think it would be appropriate to create a function `export_json_str` and `import_json_str` and use StringIO. It is also essential to maintain Python 2.7x compatibility with geomdl v5.x

On the other hand, in geomdl v6.x, that part of the code has been refactored heavily. I guess I can update all string-based functions to use StringIO without changing the type of the return value.

orbingol, 2020-08-08:
I've added `_str` functions for all exchange options in 6.x-dev. Please open an issue if anything needs to be ported back to 5.x

closed 2020-08-08 by orbingol