orbingol/NURBS-Python #77: How to move the nurbs arc?

friends110110, 2020-02-27:
I provide some specific points,  using `interpolate_curve(points, degree) ` to generate a curve, 
now I wanna modify the  start point  and end point of the specific points, and make the nurbs  curve move smoothly. How to do that ? Please provide the code, thanks

orbingol, 2020-03-01:
@friends110110 have you tried translating the control points? `linalg` module might also be helpful for that purpose.

friends110110, 2020-03-02:
@orbingol Thanks for your reply.
I have no data about the translation of the control points ,  it only provide the translation of two interpolate points  ( which are start and end points) , 
could u provide some code about translate endpoint of the arc according to the interpolate nurbs curve?

friends110110, 2020-03-04:
@orbingol could you give some advices?

orbingol, 2020-03-06:
`interpolate_curve` creates control points and knot vectors for a given degree. By default, it creates a clamped curve (or knot vector), which means start and end points of the curve are clamped to the control points polygon.

There is also a way to translate the whole curve using `operations.translate` function.

friends110110, 2020-03-07:
@orbingol now I have a nurbs curve , could you give me the examples about translate  the start point and end point  ,the the whole nurbs curve moves,  and  ignore the degree and knot vectors parameters ?

orbingol, 2020-03-08:
@friends110110 I am trying to understand question but it doesn't make sense.

The start point of a clamped curve is the 1st element in the control points array and the end point of a clamped curve is the last element in the control points array. You may use `linalg.point_translate` to move those points using a vector. Then, you need to re-evaluate the curve.

For `operations.translate`, please refer to the examples repository. You may want to take a look at examples splitting the curve by knot insertion and moving one of the split curves using a vector.

friends110110, 2020-03-09:
@friends110110 how to only add offset of the start and end points , and the other points on the nurbs curve2 move regularly automatically?

orbingol, 2020-03-10:
There is no automatic way to do that as of geomdl v5.2.9. Like I said in https://github.com/orbingol/NURBS-Python/issues/79#issuecomment-596212178, you will need to implement offsetting algorithms and adapt them to your specific conditions. There may be other ways to solve your problem but you will still need to implement them.

friends110110, 2020-03-10:
@orbingol  could u give some advices to help me to realize this function?

orbingol, 2020-03-13:
Unfortunately, offsetting keyword might be the only lead I could give to you. I am sorry. I am pretty sure that you will find several research papers on curve offsetting.

Since geomdl does not include any offsetting algorithms as of v5.2.9, I'd like to consider this issue as solved.

closed 2020-03-13 by orbingol