orbingol/NURBS-Python #78: How to find a nearest point on BSpline curve or surface?

heshijiu, 2020-03-02:
which API can be used to get the nearesr point on BSpline curve or surface

orbingol, 2020-03-06:
@heshijiu I am sorry but there is no API call included in geomdl for that purpose.

portnov, 2020-03-06:
@orbingol am I correct in assuming that there is just no better algorithm for this task than "rasterizing" / tesselating the curve / surface and then using k-d tree or BVH structure? Or maybe there is such algorithm, but it just is not implemented in the library?

orbingol, 2020-03-08:
@portnov that feels like a different question, more related to the research but not software implementations.

I am closing this issue as `geomdl` does not provide any implementations of finding closest point relative to an arbitrary point, as of today.

closed 2020-03-08 by orbingol