orbingol/NURBS-Python #79: How to generate nurbs curve according another curve?

friends110110, 2020-03-07:
How to generate nurbs curve2 according to the nurbs curve1 (using the interpolate points) , the start point and end point on the nurbs curve2 ?

[image]

orbingol, 2020-03-08:
This looks like offsetting but you seem to have some special conditions. There are no offsetting algorithms included in `geomdl` (I'd say yet, hopefully I will add in future)

friends110110, 2020-03-09:
@orbingol  I wanna to generate nurbs curve2  according to nurbs curve1,  according to nurbs curve1 and  the start point and end point by adding offset,  do you have some solutions?

orbingol, 2020-03-13:
Related to #77

closed 2020-03-13 by orbingol