orbingol/NURBS-Python #80: how to solve the derivatives of basis functions

GabrielJie, 2020-03-07:
Dear authors,

In the current version, I find the corresponding module to solve the derivatives of B-spline or NURBS. In actual, I will need the derivatives of their basis with respect to the parametric coordinates when using NURBS/B-spline in isogeometric analysis (IGA).

Hence, how to solve it. I have overlooked all modulus in the current version, and there might be not featured with this function.

With best regards,
Jie.

orbingol, 2020-03-08:
It is in the `helpers` module.

GabrielJie, 2020-03-09:
Thanks, 

I find it, and it is not shown in the documentation.

With best regards,
Jie

orbingol, 2020-03-09:
> it is not shown in the documentation.

Actually, it is: https://nurbs-python.readthedocs.io/en/latest/module_utilities.html#helpers

I am closing this issue. Feel free to reopen if you have any issues.

closed 2020-03-09 by orbingol