orbingol/NURBS-Python #86: Why interpolate_curve a new curve , its control points have so many ?

friends110110, 2020-04-01:
```
from geomdl import BSpline, multi

# Create the curve instance
from geomdl.fitting import interpolate_curve
from geomdl.visualization import VisMPL

crv = BSpline.Curve()

# Set degree
crv.degree = 2

# Set control points
crv.ctrlpts = [[1, 0], [1, 1], [0, 1]]

# Set knot vector
crv.knotvector = [0, 0, 0, 1, 1, 1]
crv.evaluate()

points = crv.evalpts
crv2 = interpolate_curve(points,2)

crv2.ctrlpts_size
```

Why crv2.ctrlpts size is 101, How to make  the control points number of crv2 curve  is 3, it should be the saime to the control points number of crv curve?

friends110110, 2020-04-07:
@orbingol could u give some reasons? Actually I wanna three control points after invoke the `interpolate_curve `  method.

orbingol, 2020-04-08:
I'd say it works as expected considering the definition of interpolation. Try setting the sample size to 3 or play with the evaluation delta to reduce the number of data points.

friends110110, 2020-04-08:
@orbingol when I use `interpolate_curve` , I change the sample size and delta by
`
crv2.sample_size=3
crv2.delta=0.9
`
It seems that the control points of crv2 do not reduce . Could u give some advices?

orbingol, 2020-04-09:
`sample_size` and `delta` are the different representations of the same variable: evaluation delta. You should set only one of them, whichever makes sense to you.

closed 2020-04-10 by orbingol

friends110110, 2020-04-13:
@orbingol when I set `crv2.delta=0.9` , it still have a lot of control points.   How to specify only three points when `interpolate_curve` a curve ?

orbingol, 2020-04-13:
I'd suggest you to check the documentation of `delta` and `sample_size` properties.

friends110110, 2020-04-14:
@orbingol  I read [Setting the evaluation delta ](https://nurbs-python.readthedocs.io/en/latest/basics.html#setting-the-evaluation-delta) , the evaluate points reduce, but the num of the control points do not reduce.

orbingol, 2020-04-15:
If you would like to use the interpolation, you need to reduce the number of evaluated points that you input to the interpolation function to control the number of control points on the interpolated curve. For instance; if you input 10 data points, interpolation function will create 10 control points.

You could also use knot removal on the interpolated curve to remove some of the control points.

friends110110, 2020-04-15:
@orbingol Thanks.
Do you know how the start point affect the whole arc curve when move the start point to another position  in some CAD(generally NURBS curve)  software ? How the control points and evaluate points change when move the  start point in the curve ?

orbingol, 2020-04-15:
> Do you know how the start point affect the whole arc curve when move the start point to another position in some CAD(generally NURBS curve) software ? How the control points and evaluate points change when move the start point in the curve ?

I'd be really happy if you would ask questions not directly related to NURBS-Python (geomdl) on the email list `nurbs-python@googlegroups.com`.

Thank you!