orbingol/NURBS-Python #89: {bezier} <-> b-spline conversion

dec1, 2020-04-30:
Hello.

Is it possible to use your library to perform (automatic) conversion of b-spline to corresponding set of bezier curves, and vice-versa? 

Also mentioned [here](https://math.stackexchange.com/questions/364054/is-it-possible-to-convert-a-b-spline-into-a-bezier-curve )

If not would it be possible to add this feature?

Thanks in Advance

orbingol, 2020-04-30:
@dec1 Would this work for you? https://nurbs-python.readthedocs.io/en/latest/visualization_splitting.html

dec1, 2020-05-01:
Hello.
Thanks.
So the actual code to do so would be 
`operations.decompose_curve()` as used eg. [here](https://github.com/orbingol/NURBS-Python_Examples/blob/c99d8cd3d20e7523694ce62f72760b260582fa11/curve2d/ex_curve02_decompose.py#L32)?

orbingol, 2020-05-01:
@dec1 yes, you are correct. You can find links to the function references on the doc page above.

dec1, 2020-05-01:
Great. Thank you very much

closed 2020-05-01 by dec1