orbingol/NURBS-Python #90: No support for rendering two or more separate figures using VisPlotly

philipptempel, 2020-05-12:
**Describe the bug**
The default filename for any new plotly html file is 'temp-plot.html` which makes rendering two or more objects into separate outputs impossible.

**To Reproduce**
Steps to reproduce the behavior:
1. Take the [3D curve example](https://github.com/orbingol/NURBS-Python_Examples/blob/master/curve3d/ex_curve3d01.py) from the examples package
2. Import plotly visualization backend `VisPlotly` instead of matplotlib `VisMPL` i.e.,
    1. L:17 `from geomdl.visualization import VisPlotly`
    1. L:40 `vis_comp = VisPlotly.VisCurve3D()`
3. Run code
4. See the same figure open in your browser twice.

**Expected Behavior**
There should be two distinct figures visible where the second figure has one more knot point.
The expected behavior can be seen when running the example code with the matplotlib visualization backend.

**Configuration:**
 - OS: Windows 10 1809-1.6 (build 17763.1158)
 - Python distribution: python.org
 - Python version: `3.7.5 (tags/v3.7.5:5c02a39a0b, Oct 15 2019, 00:11:34) [MSC v.1916 64 bit (AMD64)]`
 - geomdl install source: PyPi
 - geomdl version/branch: 5.2.9

**Additional Details (Optional)**
`VisPlotly` stores the figure filename in `VisPlotly.vconf.figure_filename` which is hard-coded to be `temp-plot.html` and does not expose a public API for changing.
The figure filename can be changed through
```python
vis = VisPlotly.VisCurve2D()
vis.vconf.figure_filename = 'new-filename'
```
but this is super cumbersome and also kind of not self explanatory.
Plotly, if used through its low-level interface, creates an in-browser file through a local http request i.e. the code
```python
import plotly.graph_objects as go
import numpy as np

x = np.arange(10)

fig = go.Figure(data=go.Scatter(x=x, y=x**2))
fig.show()
```
opens a browser window at `127.0.0.1:56712` (or some other random available port).

orbingol, 2020-05-12:
@philipptempel thanks! You can use `VisPlotly.VisConfig` class to control the output filename till it the easy method gets fixed. This was the initial approach of using visualizer modules; create a config, create a visualizer and display the output. There can be multiple visualizers with the same or separate configuration. It looks complicated, but also it gives more flexibility.

I guess I removed all the examples showing config -> visualizer -> display method from the examples repo in favor of the new and easy method.

Here is the updated `ex_curve3d01.py` example:

```
import os
from geomdl import BSpline
from geomdl import utilities
from geomdl import exchange
from geomdl.visualization import VisPlotly as vis

# Fix file path
os.chdir(os.path.dirname(os.path.realpath(__file__)))

# Create a B-Spline curve
curve = BSpline.Curve()

# Set up curve
curve.degree = 4
curve.ctrlpts = exchange.import_txt("ex_curve3d01.cpt")

# Auto-generate knot vector
curve.knotvector = utilities.generate_knot_vector(curve.degree, len(curve.ctrlpts))

# Set evaluation delta
curve.delta = 0.001

# Evaluate curve
curve.evaluate()

# NEW: Create a config and attach it to the visualizer
c1 = vis.VisConfig()
c1.figure_filename = "test.html"
v1 = vis.VisCurve3D(c1)

# Set vis module
curve.vis = v1

# Display the curve
curve.render()

# Insert a knot
u = 0.2
curve.insert_knot(u)

# NEW: Create a new config and attach it to the visualizer
c2 = vis.VisConfig()
c2.figure_filename = "test2.html"
v2 = vis.VisCurve3D(c2)

# Update vis module
curve.vis = v2

# Display the new curve
curve.render()
```

philipptempel, 2020-05-14:
Alright, I see now how this should work. And I do understand the flexibility of using a configuration object, but I feel that there is still a bit of either a misleading parameter, a misconception, or simply a misunderstanding on my side.

Using the visualization configuration, as you have shown, works for me. But, what about the `filename`-kwarg of `curve.render()`? From your example, I would assume that the visualization configuration and the `filename`-kwarg should yield the same result of having a file called `test2.html` in the end (in your latter example).
In addition, I would assume calling `curve.render(filename='some-other.html')` would override any value defined in the visualization config i.e., act as a local override.

However, that does not seem to be the case. The lines corresponding are [L203-L216](https://github.com/orbingol/NURBS-Python/blob/master/geomdl/visualization/VisPlotly.py#L203-L216) if `VisPlotly.Curve2D` which configure the filename of the plotly-generated HTML:

```python
fig_filename = kwargs.get('fig_save_as', None) # `fig_save_as` is `filename` from `Curve.render(filename='...')`
# ...
plotfn_dict = {
    'show_link': False,
    'filename': self.vconf.figure_filename,
    'image': None if fig_display else self.vconf.figure_image_format,
}
if self.vconf.no_ipython:
    plotfn_dict_extra = {
        'image_filename': self.vconf.figure_image_filename if fig_filename is None else fig_filename,
        'auto_open': fig_display,
    }
```
Of course there are more occurrences of this snippet in `VisPlotly.Curve3D`, `VisPlotly.Surface`, and `VisPlotly.Volume`.

I feel that `filename` and `image_filename` are mixed up here as `filename` is the output file's name i.e., whatever plotly writes, and `image_filename` is the filename of the snapshot you'd get when you click on the "download plot as png" (camera-icon) on the HTML file. Weirdly enough, this isn't even picked up - but that may be an issue of plotly and not geomdl.

Summarizing: `curve.render(filename='test2.html')` does not create the same result as using a visualization configuration with `conf.figure_filename = 'test2.html'`.

orbingol, 2020-05-14:
You are completely right. `curve.render(filename='myfile.html')` should save the figure to `myfile.html` as expected.

The initial design of `VisPlotly` was for plotly v2.x (or early v3.x releases) and as far as I remember, it was working at that time. I understand the confusion between `filename` and `image_filename`, but it was something related to plotly itself. There was a major refactoring in plotly v4.x and I did the minimal changes to make `VisPlotly` compatible with plotly v4.x, which seemed better than forcing people to use plotly v3.x.

The real question here is the tests. This issue should have been identified and fixed long ago. However, I had some issues with CI integration and plotly back in time I was writing the visualization module tests. Luckily, plotly separated their offline and online rendering system with the release of v4.x and I am guessing it might be easier now to write visualization tests for `VisPlotly` module.

orbingol, 2020-07-08:
This issue will be resolved with the next release of geomdl v5.x

closed 2020-07-08 by orbingol