orbingol/NURBS-Python #91: Numpy usage?

portnov, 2020-05-26:
Hello.
As far as I understood from Contributing.md, nurbs-python does not use numpy for it's core algorithms, such as evaluating nurbs curves/surfaces.
Could you please explain why such decision has been made? Maybe this library is targeting on some specific platform where numpy is not available?

The reason why I'm asking is because I've recently implemented Bezier curves in python+numpy (only evaluation, without complex algorithms such as fitting/interpolation), and it is very fast. From my experience with numpy, the performance of python+numpy code can easily be very near C++, because numpy uses SIMD instructions automatically; and then there is no need for Cython for performance improvements.

orbingol, 2020-07-30:
I think I missed this question :)

We all know that if we can vectorize the equations properly, we will take advantage of every single bit of the library and the hardware. It has been a long time but what I remember is that when we vectorize the NURBS equations, we create sparse matrices (I hope I remember it correctly). I think we add another level of complication when we go with the sparse matrices.

Another alternative is use `numpy.array` and use the good-old for loops. This doesn't take any advantage of the vectorization at all.

When it comes to computing basis functions, another non-vectorizable thing pops up: recursion. That's why most libraries support only quadratic and cubic B-splines. What if you would like to use higher-order B-splines?

There are lots of places that I could have used numpy, but it might complicate the things on the sole purpose of this library: compute NURBS

If you would like to know why Cython, it was just an experiment and it somehow worked. Now, I can say "use Cython if you want performance" when users ask for it. When you look at the perspective of "there is no need", I'd easily say there is no need for Python as long as you have C and we jump into an uncessary infinite loop.

Well, of course, you can say "you should have done that, you should have done this" but this is my decision, very much like your decision to use anything you want. I could have asked you the same question, replacing numpy with x86 assembly language :) You know, telling me you don't know x86 assembly language wouldn't end the conversation :)

portnov, 2020-07-30:
Well, I can now say that it is possible to implement nurbs with numpy :) You can check https://github.com/nortikin/sverchok/blob/native_nurbs/utils/curve/nurbs.py and https://github.com/nortikin/sverchok/blob/native_nurbs/utils/surface/nurbs.py . Since you're familiar with nurbs maths, I hope you will find the code readable enough (though I intend to document it better). In my cases, numpy-based implementation was about 10x faster than geomdl + cython when evaluating curves / surfaces in many points at once (about 200 points). (i have to say I did benchmarks inside Blender, so it can appear that cython just can't function properly when running inside blender, I didn't check). Unfortunately I don't currently have a variant of code that can run without blender+sverchok... Maybe I'll do it later.

orbingol, 2020-07-30:
@portnov it was always possible to implement nurbs with numpy :) I am not even arguing that. It is just a matter of decision.

Also, it is always good to give some researchers opportunity to build faster libraries and cite `geomdl` as slow :) I wouldn't mind that :)

Regarding cython on Blender, you might be correct. Blender includes an embedded version of Python as long as I remember. Years back, we had been having issues with numpy installed on Blender python and some other C-based python modules. On the other hand, Cython compilation was never intended to be a major feature. I'd always say that, if you need performance, use CUDA or OpenCL and a compiled language, preferably C. I've experimented with Numba and it is a good alternative too, but not the easiest when you encounter problems.

portnov, 2020-07-31:
A bright side of Python is that you always can use C/C++ libraries, with numpy being just one of them. So in fact when we say "implement something in numpy" it actually means "implement in C++". And numpy uses SIMD instructions when it can. I did some benchmarks some time ago and I saw that python+numpy code can be actually faster than hand-written C++ code, just because numpy uses SIMD by default, while in C++ you have to think about how to make compiler use them (or select an appropriate library). So "if python is slow, use C++" is not totally true; I'd say "if python is slow, first try numpy, and only it's too slow try C++".

orbingol, 2020-07-31:
> A bright side of Python is that you always can use C/C++ libraries

Sure! I appreciate the work of pybind11 developers, so that I can convert whatever I need to Python.

> So in fact when we say "implement something in numpy" it actually means "implement in C++".

I disagree with this statement. It is comparing apples and oranges here. However, I agree with the following.

> python+numpy code can be actually faster than hand-written C++ code

However, I'd say it depends. It is not an easy job to code properly in C++. Python is more forgiving in that sense.

> numpy uses SIMD by default

This seems to be a pretty new enhancement: https://numpy.org/neps/nep-0038-SIMD-optimizations.html

I wasn't aware of it, good to know, thanks!

> So "if python is slow, use C++" is not totally true; I'd say "if python is slow, first try numpy, and only it's too slow try C++".

There is a misunderstanding here. I am not trying to bury Python or I am not a numpy hater. I believe it is not even appropriate to bury Python or numpy under a Python code repository. I appreciate numpy developers and they are doing a great job. I am just saying compiled languages are better, if you need performance. There are pointer tricks, and other things that you can apply. However, the developer needs to spend more time to know and apply those tricks. It just doesn't come super-fast out of the box. However, python + some very nice C-based modules can give you a nice performance without any knowledge of performance tricks and help you find your way when you have a limited budget (money, time, staff, and so)

If we get back to the question of why Cython, I hope it is all clear for you. There are no plans to convert everything in this repository to numpy, at least in near future (even though I convert, I might not be able to open-source it). Cython will probably stay as it is, but I am planning to add `geomdl.core` to the Anaconda Cloud repository.

I've used Numba with so many different project and I was able to get even better performance than Fortran/C/C++ code (yes, we did that competition once or twice, including the CUDA competition). There is also a very nice comparison on their paper. There might be an option to add Numba code to the repository in future but I can't see when at the moment.

I am closing the ticket for now. Thanks for all the information @portnov!

closed 2020-07-31 by orbingol