orbingol/NURBS-Python #92: Wrong number of surface points generated

pablocael, 2020-05-27:
**Describe the bug**
Just generate surface using this code below
py```
 size = gridItem.GetSize()
        # Set degrees
        self._surf.degree_u = 2
        self._surf.degree_v = 2

        controlPoints = []
        items = gridItem.GetItems()
        for row in items:
            for item in row:
                pos = gridPos + item.pos()
                controlPoints.append([pos.x(), pos.y(), 0])

        self._surf.set_ctrlpts(controlPoints, size[0], size[1])
        self._surf.knotvector_u = geomdl.knotvector.generate(self._surf.degree_u, size[0])
        self._surf.knotvector_v = geomdl.knotvector.generate(self._surf.degree_v, size[1])

        gridPixelSize = gridItem.GetPixelSize()
        self._surf.sample_size_u = gridPixelSize[0]
        self._surf.sample_size_v = gridPixelSize[1]
```
I have 460 x 250 grid points. When I set _surf.sample_size_u to 460 and _surf.sample_size_v to 250,
I print their values and they are 459 and 250 respectivelly.  ```self._surf.evalpts``` also retrieves
```114750```points which is actually 459x250 and not 460x250. 

The control points used are attached in the file below. Just create the same surface using exact same controlpoints. 

Why is degree_u being set to 1 less then what I actually set?

Thanks!

[controlpoints_list.txt](https://github.com/orbingol/NURBS-Python/files/4688814/controlpoints_list.txt)

orbingol, 2020-05-27:
I don't think `degree_u` is changing.

The system mainly works with `delta` property and `sample_size` is a convenience property that the input is being converted into a floating point number between 0 and 1, which is the `delta` value. What you are currently observing is a floating point error. I'd suggest using `delta` property directly to set the sample sizing.

pablocael, 2020-05-27:
First of all thanks for the quuck reply
Sorry, i meant sample_size_u is changing. Degrees are not changing, sorry for the confusion.
Im setting samples size u and v but they are not staying in the value I set, but 1 unit below.
It really looks like rounding error. I guess I can try using delta explicitly.

Nevertheless I think this behavior can be sometimes odd, cant it? In my case its breaking the uv texture calculation.
I guess event if its a convenience function it should preserve the set sample size or maybe something on the doc that warns about that rounding possibility?

Thanks a lot

orbingol, 2020-05-28:
> Nevertheless I think this behavior can be sometimes odd, cant it? In my case its breaking the uv texture calculation.
> I guess event if its a convenience function it should preserve the set sample size or maybe something on the doc that warns about that rounding possibility?

You are right, documentation is not that clear on how `sample_size` works.

pablocael, 2020-05-30:
Hi there, I have tried setting delta_u and delta_v directly, but result remains the same:
I've set:      
```
self._surf.delta_u = 1.0 / (gridPixelSize[1]-1)
self._surf.delta_v = 1.0 / (gridPixelSize[0]-1)
```
where gridPixelSize = (460, 250)
But number of surface points generated is 1 less along u:

number of surface points = len(self._surf.evalpts) => 114750, gridPixelSize = (460, 250)

114750 = 459 x 250

Its always missing 1. If I query _surf.delta_u and _surf.delta_v, I get exactly what was set:

1.0 / 2 and 1.0 / 460, to the 11 decimal place correct:

delta_u = 0.004016064257028112
delta_v = 0.002178649237472767

So 249 * delta_u = 1 and 459 * delta_v = 1

Which mean I will have 250 and 450 points respectivelly, assuming points from 0 to 1.0/delta inclusive.

I cannot make it generate the right number of points.

pablocael, 2020-05-30:
I can reproduce the problem even in the example from the doc:

from geomdl import BSpline

# Create a BSpline surface instance (Bezier surface)
surf = BSpline.Surface()

# Set degrees
surf.degree_u = 3
surf.degree_v = 2

# Set control points
control_points = [[0, 0, 0], [0, 4, 0], [0, 8, -3],
                  [2, 0, 6], [2, 4, 0], [2, 8, 0],
                  [4, 0, 0], [4, 4, 0], [4, 8, 3],
                  [6, 0, 0], [6, 4, -3], [6, 8, 0]]
surf.set_ctrlpts(control_points, 4, 3)

# Set knot vectors
surf.knotvector_u = [0, 0, 0, 0, 1, 1, 1, 1]
surf.knotvector_v = [0, 0, 0, 1, 1, 1]

# Set evaluation delta (control the number of surface points)
surf.delta_u = 1.0/249
surf.delta_v = 1.0/459

# Get surface points (the surface will be automatically evaluated)
surface_points = surf.evalpts

print(len(surface_points)) # 114750

orbingol, 2020-05-30:
I am guessing there should be a problem with the `linspace` implementation in the `linalg` module.

pablocael, 2020-05-30:
Thanks for the quick reply.

I have pulled version 6.x-dev to check the fix. I spotted an issue when evaluating the surface now:

`
  File "main.py", line 839, in updateRender
    surface_points = self._surface.Eval()
  File "main.py", line 342, in Eval
    return self._surf.evalpts
  File "/usr/local/lib/python3.7/site-packages/geomdl/abstract.py", line 63, in evalpts
    self.evaluate()
  File "/usr/local/lib/python3.7/site-packages/geomdl/BSpline.py", line 257, in evaluate
    super(Surface, self).evaluate(**kwargs)
  File "/usr/local/lib/python3.7/site-packages/geomdl/abstract.py", line 638, in evaluate
    self.check_variables()
  File "/usr/local/lib/python3.7/site-packages/geomdl/abstract.py", line 609, in check_variables
    validate_degree_value(self._attribs[i], self._degree[i])
  File "/usr/local/lib/python3.7/site-packages/geomdl/abstract.py", line 708, in validate_degree_value
    raise GeomdlError("Degree " + str(key) + " should be bigger than zero")
geomdl.base.GeomdlError: GEOMDL ERROR: Degree u should be bigger than zero
`

this happens even with the simple example with the site.

I am totally aware this is a unstable development version, but I want to try. This issue might be something caused by some recent changes? I didnt dig into the code to check (sorry about that).

Thanks in advance.

orbingol, 2020-05-30:
`6.x-dev` is still under development and its API is different. I would NOT recommend using it right now.

pablocael, 2020-05-31:
I see, thanks a lot for the support.

orbingol, 2020-08-02:
I've pushed a fix on `5.x-dev` branch. Hopefully, it will fix this issue. I'll need to test it a bit more to make sure the fix working fine.

orbingol, 2020-08-07:
This issue has been resolved. Please install geomdl v5.3.0

closed 2020-08-07 by orbingol