orbingol/NURBS-Python #93: How to move the scaled point and keep the shape of the curve?

friends110110, 2020-05-28:
For a curve, I wanna move the start and end point to new position,  and then how to keep the shape of the curve ?

[image]

From the points which circulated on the yellow curve,  move them to red curve position which circulated on the red curve, how to keep the shape of the curve on the red curve ?

friends110110, 2020-05-28:
@orbingol @cshorler @heczis @Nodli Could you help me to solve this problem? move the start and end point of a curve to another position, how to keep the shape of new curve ?

friends110110, 2020-05-29:
@orbingol could you help me to solve this problem?
Thank you

orbingol, 2020-07-30:
@friends110110 this is not directly related to geomdl, like a bug or something else. Due to time constraints, I can't help you. I am closing this issue.

closed 2020-07-30 by orbingol