orbingol/NURBS-Python #95: How to get fitting coefficients?

Jshcht, 2020-06-08:
**Is your feature request related to a problem? Please describe.**
Thanks for the convenient package. I wonder if I can extract the coefficients of a curve or surface fitting.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

orbingol, 2020-06-09:
@Jshcht There are some functions for this purpose in `fitting.py`. I'd suggest you looking at line 509. They are actually documented but not exported to the online documentation.

Jshcht, 2020-06-10:
Thanks a lot for your kind rely and providing such a convenient package. I will check it out.

orbingol, 2020-06-16:
I am closing this issue and marking it as resolved.

closed 2020-06-16 by orbingol