orbingol/NURBS-Python #96: Accessing the trim 'negative'

j-polden, 2020-06-15:
Hi There, really impressed with this library so far!

I have been using the trim functionality to cut holes into a generated surface, however I am trying to export the trimmed section of this surface. Does anyone have any tips how we could access this? 

To explain a little better, please see the picture below. I am able to trim a polygon from the surface, after which I am left with the larger surface, but not the trimmed section/patch. Are we able to access this trimmed patch, and perhaps plot it? Source code for this example is supplied below.

[image]
 
Sample code:
> import os
from matplotlib import cm
from geomdl import BSpline
from geomdl import NURBS
from geomdl import exchange
from geomdl import tessellate
from geomdl.visualization import VisMPL as vis
from geomdl import knotvector
from geomdl import trimming
os.chdir(os.path.dirname(os.path.realpath(\_\_file\_\_)))
surf = BSpline.Surface() 
surf.degree_u = 3
surf.degree_v = 3  
ctrl_pts = [[[-25,-25,-10],[-25,-15,-5],[-25,-5,0],[-25,5,0],[-25,15,-5],[-25,25,-10]],
[[-15,-25,-8],[-15,-15,-4],[-15,-5,-4],[-15,5,-4],[-15,15,-4],[-15,25,-8]],
[[-5,-25,-5],[-5,-15,-3],[-5,-5,-8],[-5,5,-8],[-5,15,-3],[-5,25,-5]],
[[5,-25,-3],[5,-15,-2],[5,-5,-8],[5,5,-8],[5,15,-2],[5,25,-3]],
[[15,-25,-8],[15,-15,-4],[15,-5,-4],[15,5,-4],[15,15,-4],[15,25,-8]],
[[25,-25,-10],[25,-15,-5],[25,-5,2],[25,5,2],[25,15,-5],[25,25,-10]]] 
surf.ctrlpts2d = ctrl_pts 
surf.knotvector_u = [0.0, 0.0, 0.0, 0.0, 1.0, 2.0, 3.0, 3.0, 3.0, 3.0]
surf.knotvector_v = [0.0, 0.0, 0.0, 0.0, 1.0, 2.0, 3.0, 3.0, 3.0, 3.0]
surf.sample_size = 20
surf.tessellator = tessellate.TrimTessellate()
visconf = vis.VisConfig(ctrlpts=False, legend=False)
surf.vis = vis.VisSurface(config=visconf)
tcrv = BSpline.Curve()
tcrv.degree = 1
tcrv.ctrlpts = [[0.2,0.3], [0.5,0.3], [0.7,0.1],[0.7,0.8],[0.2,0.3]]
tcrv.knotvector = knotvector.generate(1, 5)
trim_curves = [tcrv]
surf.trims = trim_curves
surf.render(colormap=cm.copper)

orbingol, 2020-06-16:
Yes, there is an experimental and not very well documented configuration variable that trims the reverse, independent of the direction of the trim curve control points.

Setting `tcrv.opt = ["reversed", 1]` would do the trick, but just a reminder, it is still experimental.

j-polden, 2020-06-16:
Amazing! Thanks :) [image]

closed 2020-06-16 by j-polden