orbingol/NURBS-Python #98: Bspline .normal does not report correct vector. Z is always 0.0

Burtski, 2020-06-26:
**Describe the bug**
When extracting the normal vector to a Bspline the returned vector is incorrect.
The Z value is always 0.00

(actual code attached)
CenterCurve = BSpline.Curve()
CenterCurve.degree = CenterLineSpline['Dimension']
CenterCurve.ctrlpts = CenterLineSpline['ControlPoints']
CenterCurve.knotvector = CenterLineSpline['KnotPoints']
CenterCurve.reverse()
print(CenterCurve.normal(0.5002508115019416))

**Result**
((114.33887671036268, 306.5311505857961, -28.2751622397278), (-0.7484488832937752, 0.6631924826898302, 0.0))

**Expected Behavior**
Normal Vector should be approximately  [-0.462917,  0.006187,  -0.886380]

For any position point on the spline Z will be reported 0.00

**To Reproduce**
Code with data attached:

**Configuration:**
Windows 10,
Python 3.7.4
Idle
geomdl 5.2.9
geomdl inatlled via pip3

**Comment**
If I am not using this correctly , please explain.

[Geomdlbug.py.txt](https://github.com/orbingol/NURBS-Python/files/4838510/Geomdlbug.py.txt)

orbingol, 2020-06-26:
I have added the following for testing:

```
from geomdl import linalg

pt_t, vc_t = CenterCurve.tangent(0.5002508115019416)
pt_n, vc_n = CenterCurve.normal(0.5002508115019416)

angle = linalg.vector_angle_between(vc_t, vc_n)
print("Computed angle:", angle)
angle = linalg.vector_angle_between(vc_t,  [-0.462917, 0.006187, -0.886380])
print("Expected angle:", angle)
```

Results:

> Computed angle: 90.00000000000001
> Expected angle: 90.00000664819953

Burtski, 2020-06-29:
Perhaps I do not understand what the .normal is telling me.
I was under the impression that the normal would 'point' to the center of curvature (assuming there is curvature at this point ).

If I interpret this correctly, the .normal my return any vector that is on the plane that is normal to the tangent.
There are therefore an infinite number of solutions for the .normal.

Will the chosen solution always be the one where z=0.0?

If this is the case, how do I withdraw my bug report?

Thank you.

orbingol, 2020-06-29:
> I was under the impression that the normal would 'point' to the center of curvature (assuming there is curvature at this point ).

That would be the curvature vector, right?

> If I interpret this correctly, the .normal may return any vector that is on the plane that is normal to the tangent.
> There are therefore an infinite number of solutions for the .normal.

Actually, "curve normal" is not a very well defined operation. We have discussed it here on Github, as well as in person. Whoever I talked explained it in a different way. Therefore, the normal computation has been updated to compute the normal to the tangent vector, which is also consistent with the most geometry references.
 
> Will the chosen solution always be the one where z=0.0?

It shouldn't be. On the other hand, you are not restricted to the computations in geomdl. You can compute the derivatives via `Curve.derivatives` and use this information to compute the vectors you need.

> If this is the case, how do I withdraw my bug report?

Just close it at the bottom.

Burtski, 2020-06-29:
Thank you for your clarification.

closed 2020-06-29 by Burtski