orbingol/NURBS-Python #99: Nurbs equally spaced points

andreabonacini, 2020-07-03:
Hi, 
i'm using the Nurbs.curve class and i would like to get a set of equally spaced evaluated points over the curve, possibly as a function of an imposed delta (or sample size). Is it possibile?
Regards,
Andrea

orbingol, 2020-07-04:
Evaluated points are equally spaced on the parametric space (UVW) where the evaluation happens. It might or might not be equally spaced on the XYZ space.

andreabonacini, 2020-07-06:
So, is there a function inside the NURBS library in order to compute equally spaced samples on the XYZ space? The input of the function i would like to be progress step previewsly defined in the XYZ space

ebo, 2020-07-27:
@andreabonacini, are you referring to arc-length parameterization of the NURB?  ie. if you have a curve that is 6" long, and you evaluate it a 1" that it is 1/6'th of the way along the curve, or are you referring to something else?

andreabonacini, 2020-07-28:
I was referring to if there was a method that takes as input a constant velocity (or a progress step) in order to obtain in output the parameter values that allow to have equally spaced points (cartesian coordinate) along the NURBS. Obviusly the parametrization u would not be equally spaced. 

Anyway, for that purpose i successfully implemented my own algorithm of a constant feed rate. This solved the problem for me.

ebo, 2020-07-28:
I would be interested in looking at it if you would be willing to 
share.

On Jul 28 2020 7:22 AM, Andrea Bonacini wrote:
> I was referring to if there was a method that takes as input a
> constant velocity (or a progress step) in order to obtain in output
> the parameter values that allow to have equally spaced points [...]

orbingol, 2020-07-28:
> Anyway, for that purpose i successfully implemented my own algorithm of a constant feed rate. This solved the problem for me.

I think we can close this issue.

closed 2020-07-28 by orbingol