orbingol/NURBS-Python #101: the automated tests appear to all be broken

ebo, 2020-07-04:
While evaluating NURBS-Python as a starting point for a numba project, I ran nosetest to start from the examples.  I consistently found from 100 to over 200 broken tests in the test suite (depending on the branch).  

What is the most stable branch that I should start from?

Are the automated test suite maintained?

Steps to reproduce:

  * start with a clean Anaconda install
  * install nose
  * run 'nosetests .'

results in 100's of broken tests.

orbingol, 2020-07-05:
I am not sure what is the problem with `nose` but all automated tests are passing (please refer to CI logs/configuration for `5.x-dev` branch for details). I use `pytest` ([docs](https://nurbs-python.readthedocs.io/en/latest/install.html#testing)). For the branch information to work with, please refer to the [wiki](https://github.com/orbingol/NURBS-Python/wiki). Current stable version is 5.x series.

Just in case, I've dropped support for Python v3.4 and pypy3.5 for `5.x-dev` branch and Python 2.7, Python 3.4 and pypy3.5 for `6.x-dev` branch.

ebo, 2020-07-05:
I have a little time at the moment to test the numba.jit support if you can release them now -- it would be less work for me writing tests than starting a branch from scratch...

As a note, my comment about the tests broken turned out to be the pypy version issue mentioned in the other bug report.  On my install using pytests also breaks with the error:

=================================================================================== ERRORS ====================================================================================
________________________________________________________________ ERROR collecting tests/test_visualization.py _________________________________________________________________
ImportError while importing test module '/home/ebo/save_all/gtd/projects/cnc/RPi/ref/NURBS-Python/tests/test_visualization.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests/test_visualization.py:15: in <module>
    import matplotlib
/cm/dropbox/anaconda/envs/pypy/site-packages/matplotlib/__init__.py:207: in <module>
    _check_versions()
/cm/dropbox/anaconda/envs/pypy/site-packages/matplotlib/__init__.py:192: in _check_versions
    from . import ft2font
E   ImportError: cannot import name 'ft2font'
=========================================================================== short test summary info ===========================================================================

This is still a system issue, but broken all the same.  I will see if I can either get pypy to work, or look into getting pytest to finish testing without a critical fail.

closed 2020-07-05 by orbingol