orbingol/NURBS-Python #102: ft2font broken for pypy3 support

ebo, 2020-07-04:
Has anyone gotten pypy3 working for VisMPL visualization support?

I am not sure you would consider this a NURBS-Python bug, but I have been unable to resolve ft2font / matplotlib install in Anaconda.  Does anyone have suggestions?

To Replicate:
  * start with a clean Anaconda environment
  * install pypy
  * run any examples which include VisMPL,  or VisVTK

orbingol, 2020-07-05:
Just FYI, it was working with pypy3.5 and Travis CI still has the logs for it. I've switched the base image to `xenial` on Travis. It seems Travis CI's `xenial` image comes with pypy3.6 and it fails to install `numpy` or `matplotlib` for some reason. 

It is not related to geomdl, but I am leaving the issue open. If someone knows how to solve it, I could add pypy3.6 tests back to the CI. I just have no time to work on it.

ebo, 2020-07-05:
Thank you.  After posting I realized that the issue was more of a system problem than NURBS-Python(_Examples), but could not figure out how to get it working yet so I had not posted a reply.  As a note, the stable versions of python which are installed on my Gentoo linux box and Anaconda are all currently 3.7 and 3.8 is on the horizon.  Interestingly enough pypy3 --version is returning:

    Python 3.6.9 (?, Apr 10 2020, 19:47:05)
    [PyPy 7.3.1 with GCC 7.3.0

And the package installed is pypy3.6 with version and channel information:

   # Name                    Version                   Build  Channel
    pypy3.6                   7.3.1                h3e02ecb_1    conda-forge

Could it be an issue with pypy3 being created with a different version of python than is installed on the machine?  I have seen issues with stuff like this in the past.  Just a thought...

ebo, 2020-07-05:
as a note, I have posted bug reports with the pypy folks to make them aware of the issue.

orbingol, 2020-07-23:
Closed as explained in #105

closed 2020-07-23 by orbingol