orbingol/NURBS-Python #103: numba support request

ebo, 2020-07-05:
I have read a couple of references to initial efforts to provide numba support to NURBS-Python.  If these repositories can be shared I have a little time I can spend on helping to get it, the test suites, and timing tests running.

orbingol, 2020-07-23:
I am sorry but sharing might not be possible due to some limitations with a publication. Maybe later but not soon.

closed 2020-07-23 by orbingol

ebo, 2020-07-23:
On Jul 22 2020 11:55 PM, Onur R. Bingol wrote:
> I am sorry but sharing might not be possible due to some limitations
> with a publication. Maybe later but not soon.

understand.  Please email me when/if it is released.  As a note, I 
would be willing to sign an NDA, but I have worked with academics for 
decades and understand the need to publish first...

ebo, 2021-03-27:
Just checking in 9 months later.  Any news?