orbingol/NURBS-Python #105: reenable pypy support

ebo, 2020-07-22:
turns out that things have stabilized a bit in the last month, and pypy now seems to play nice again.  I was able to get it working and running with the NURBS-Python_examples yesterday.  As a note, I removed  "strict" from the channel flags.

orbingol, 2020-07-23:
@ebo thanks!

I am closing this issue as geomdl core should work on any python distribution. There is nothing to enable, other than the tests. Regarding the tests, I've switched to miniconda to reduce the effort of enabling plotly and vtk offscreen rendering. I am not sure if conda allows creating a pypy environment, but I'll check it out later (not a priority task due to time issues)

Issues with numpy and others are related to pypy itself.

closed 2020-07-23 by orbingol

ebo, 2020-07-23:
On Jul 22 2020 11:53 PM, Onur R. Bingol wrote:
> @ebo thanks!
>
> I am closing this issue as geomdl core should work on any python
> distribution. There is nothing to enable, other than the tests.
> Regarding the tests, I've switched to miniconda to [...]

@orbingol, you are most welcome.

switching to miniconda makes sense for the core package.

Conda, does allow for the creation of a pypy environment, that is how I 
did it.  I will have to go through the steps to make sure I remembered 
everything, but I think the last time around it was as simple as:

   conda create --name pyp
   conda install geomdl

but as I recall not all of the requirements installed and I had to add 
some things piecemeal.

As a note, I will be in and out of the field for the next couple of 
weeks and will make an effort to go through these steps and document it. 
If you do not get to it before me, or if I do not get to this within 
the next month, please feel free to ping me and I will do my best to 
write up a clean how to from scratch to getting pypy working with 
geomdl.