orbingol/NURBS-Python #106: curve-length approximation

ebo, 2020-07-27:
I have a question or two about the implementation of the curve lengths and fairing.

If I understand the documentation correctly, you use the Euclidean distance summed across some arbitrary evaluation points, is that correct?  

I was reading some papers on using NURBS for CNC machining and came across  one [1] that uses an adaptive quadrature method based on Simpson׳s rule [2] for greater accuracy.  

Also, do you have smoothing (fairing) functions where I can minimize high order derivatives (jerk, or maybe up the ladder to snap, crackle, or pop).
 

[1] Xu Du, Jie Huang, Li-Min Zhu, A complete S-shape feed rate scheduling approach for NURBS interpolator, Journal of Computational Design and Engineering, Volume 2, Issue 4, October 2015, Pages 206–217, https://doi.org/10.1016/j.jcde.2015.06.004

[2] Lei W.T., Sung M.P., Lin L.Y., Huang J.J. Fast real-time NURBS path interpolation for CNC machine tools. International Journal of Machine Tools and Manufacture 2007;47(10)1530–41.

orbingol, 2020-07-27:
> If I understand the documentation correctly, you use the Euclidean distance summed across some arbitrary evaluation points, is that correct?

Yes, and this is not a 100% correct method to compute curve length. Therefore, this function has been removed in v6.x.

> Also, do you have smoothing (fairing) functions where I can minimize high order derivatives (jerk, or maybe up the ladder to snap, crackle, or pop).

I don't think this is related to NURBS directly. I've heard that some researchers are using external libraries for this purpose.

ebo, 2020-07-27:
On Jul 27 2020 1:04 PM, Onur R. Bingol wrote:
>> If I understand the documentation correctly, you use the Euclidean 
>> distance summed across some arbitrary evaluation points, is that 
>> correct?
>
> Yes, and this is not a 100% correct method to compute curve length.
> Therefore, this function has been removed in v6.x.

What will replace it in 6.X?  Also, when is 6.X expected to be 
released?

>> Also, do you have smoothing (fairing) functions where I can minimize 
>> high order derivatives (jerk, or maybe up the ladder to snap, crackle, 
>> or pop).
>
> I don't think this is related to NURBS directly. I've heard that some
> researchers are using external libraries for this purpose.

I will have to review some of my references and see if I can either 
articulate it better, or possibly write those functions.

orbingol, 2020-07-27:
>  when is 6.X expected to be released?

As soon as possible if I can find more time to work on it.

ebo, 2020-07-27:
On Jul 27 2020 1:58 PM, Onur R. Bingol wrote:
>>  when is 6.X expected to be released?
>
> As soon as possible if I can find more time to work on it.

Understand.  One of the infinite round-to-its...  I should show you my 
list ;-)

Best of luck and success, and here is to your health!

orbingol, 2020-07-28:
@ebo thanks!

The story behind 6.x was just to separate visualization component from the core library and make the classes a little bit thinner. It was supposed to be released on January 2020 :) Then, things happened involving very very big changes, COVID-19 happened (which shouldn't be an excuse for software people, but...) and with some other personal reasons, I've started to find less free time for geomdl. The software side has been all volunteer work from the beginning, whenever I find free time.

My latest plan is keeping 6.x in beta for a while, even after I release the first version. It is not a replacement for everything-included 5.x, but mostly I'd like to keep it to be used as a part of a bigger application. Considering some requests I've received months ago, developing 6.x as a multi-component library makes more sense for the larger audience.

ebo, 2020-07-28:
@Onur, you are welcome, and our time is what it is.   I do appreciate 
having what you and others give.

multi-component libraries sounds good.  It can simplify things once in 
place.

Best of luck with everying (including finding more free time ;-)

On Jul 28 2020 8:42 AM, Onur R. Bingol wrote:
> @ebo thanks!
>
> The story behind 6.x was just to separate visualization component
> from the core library and make the classes a little bit thinner. It
> was supposed to be released on January 2020 :) [...]