orbingol/NURBS-Python #113: Unexpected output using VisSurface

ghareth, 2020-11-16:
**Describe the bug**
The visualisation produced in matplotlib is not as expected.  The surfaces extend out from the original model. 

**To Reproduce**

I am  visualising the rhino logo example file from McNeel converted using `on2json`. 

```
from pathlib import Path
from geomdl import multi
from geomdl import exchange
from geomdl.visualization import VisMPL

rhino_logo=Path('Rhino Logo.json')
data = exchange.import_json(str(rhino_logo))
surf = multi.SurfaceContainer(data)

vis_config = VisOCC.VisConfig(ctrlpts=False,
                              trims=False,
                              evalpts=True,
                              figure_size=[9, 8])

vis_comp = VisOCC.VisSurfWireframe(config=vis_config)  # for Figure_1 below
# vis_comp = VisOCC.VisSurface(config=vis_config) # for Figure_2 below
# vis_comp = VisOCC.VisSurface_amended(config=vis_config)  # for Figure_3 below

surf.vis = vis_comp
surf.render()
```

**Expected Behavior**

Figure_1 below is as expected using VisSurfWireframe
Figure_2 below shows the unexpected output when using VisSurface
Figure_3 below shows expected output using an amended VisSurface class

`__init__` method amended as follows:
```
self._module_config['evalpts'] = "quads"
```

`render` method amended as follows:
```
            # Plot evaluated points
            if plot['type'] == 'evalpts' and self.vconf.display_evalpts:
                # Use internal triangulation algorithm instead of Qhull (MPL default)
                verts = plot['ptsarr'][0]
                tris = plot['ptsarr'][1]
                # Extract zero-indexed vertex number list
                tri_idxs = [tri.data[0:3] for tri in tris] + [[tri.data[0]] + tri.data[2:4] for tri in tris]
```

**Configuration:**
 - OS: MacOS Mojave
 - Python distribution: Anaconda
 - Python version: 3.8
 - geomdl install source: installed via github clone using cython
 - geomdl version/branch: version 5.3 commit 48092978a102dfb0ff470225ca8bd2771fbb93d2

**Screenshots**

[image]
[image]
[image]

orbingol, 2020-11-17:
I have some questions:

1. I see that you are using `VisOCC` class in your example. Is that the same with `VisMPL`? Did you make any changes? If so, would you please share?
2. Could you also share the link to the model that you used (or attach here)?
3. What happens if you use triangles instead of quads? (I see that `VisSurface` uses triangles by default for evaluated points)

ghareth, 2020-11-17:
1. Sorry, VisOCC is a typo - It should read VisMPL.  I have started writing a visualisation module for PythonOCC but it is very basic.

2A. File is at https://files.mcneel.com/TEST/Rhino%20Logo.3dm
2B. JSON conversion is attached. 
[Rhino Logo.json.zip](https://github.com/orbingol/NURBS-Python/files/5553823/Rhino.Logo.json.zip)

3. If I use triangles instead of quads, the output in figure_2 is produced.

Thanks

orbingol, 2020-11-17:
I understand. Thanks for the report @ghareth. Just a small question, something is not clear to me. You wrote

> geomdl install source: installed via github clone using cython

in your report. Are you using `geomdl.core`, which is compiled via cython or `geomdl`, pure python version? Both should work fine in theory, but just wanted to make sure.

ghareth, 2020-11-17:
Ok. Didn't realise that I had to import via geomdl.core. 
I've been using Pure Python. I've now tested it against the cythonised version and there is no difference (apart from being much faster!)

orbingol, 2021-01-21:
Sorry for the late response. I've tested this a while ago. The surfaces are trimmed and they need to be tessellated with `TrimTessellate`. However, it looks like there is also a problem with the `TrimTessellator` class and it doesn't seem to output the correct result for some reason.

I have been looking into replacing `TrimTessellator` class for a while (it is slow and sometimes incorrect) and I'll try to take a look after fixing some issues with the `6.x-dev` branch.